bmax121/APatch #313: Mount failed and SU not found during module installation

daifiyum, 2024-03-21:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

安装[youtube-revanced-magisk-v19.04.37-all.zip](https://github.com/j-hc/revanced-magisk-module/releases/download/20220933/youtube-revanced-magisk-v19.04.37-all.zip)模块时，出现``Mount failed 和 SU not found``

### Reproduce method/复现方法

1. 安装最新版Apatch
2. 刷入[youtube-revanced-magisk-v19.04.37-all.zip](https://github.com/j-hc/revanced-magisk-module/releases/download/20220933/youtube-revanced-magisk-v19.04.37-all.zip)
3. 观察模块安装日志，可发现报错

### Expected behavior/预期行为

Mount failed and SU not found during module installation

### Actual behaviour /实际行为

Mount failed and SU not found during module installation

### Screenshots/截图

[image]

### Logs/日志

[APatch_bugreport_2024-03-21_07_43.tar.gz](https://github.com/bmax121/APatch/files/14675858/APatch_bugreport_2024-03-21_07_43.tar.gz)

### Device Name/设备名称

k30 pro

### OS Version/系统版本

12

### APatch Version/APatch 版本

10570

### Kernel Version/内核版本

4.19.273

### KernelPatch Version/KernelPatch 版本

0.10.4

### Additional context/其他信息

[youtube-revanced-magisk-v19.04.37-all.zip](https://github.com/j-hc/revanced-magisk-module/releases/download/20220933/youtube-revanced-magisk-v19.04.37-all.zip)

daifiyum, 2024-03-21:
补充：当使用 kernelsu 安装此模块时，未出现任何报错

daifiyum, 2024-03-21:
使用nsenter替换su，问题解决
https://github.com/daifiyum/revanced-magisk-module/commit/b157c0af68b410226e5a58f1f7f49d03ed944125

daifiyum, 2024-03-21:
现在的kernelsu可以直接使用su，似乎无需nsenter，我也不是太懂，希望作者能看看

pomelohan, 2024-03-23:
有没有试过设置里面的全局命名空间挂载？

bmax121, 2024-03-23:
模块安装的时候本来就是 root， uid 为 0，但是 uid 为 0 其实是没有使用 su 的权限的，用不了 su

daifiyum, 2024-03-23:
> 有没有试过设置里面的全局命名空间挂载？

后面发现了这个选项，我试了一下，还是不行
具体操作：设置里面开启，重启，刷入，还是报错

pomelohan, 2024-03-24:
有没有试过直接去掉 su -m -c

daifiyum, 2024-03-24:
> 有没有试过直接去掉 su -m -c

去掉后测试没问题，但我不确定这对kernelsu和magisk有没有问题

pomelohan, 2024-03-24:
> > 有没有试过直接去掉 su -m -c
> 
> 去掉后测试没问题，但我不确定这对kernelsu和magisk有没有问题

但这看起来并不是 APatch 的问题了，就像 bmax 所讲的那样，

> 模块安装的时候本来就是 root

因为已经是最高权限了，su 是不存在的。如果有必要可以让模块作者改进。

closed 2024-03-24 by pomelohan

daifiyum, 2024-03-24:
> > > 有没有试过直接去掉 su -m -c
> > 
> > 去掉后测试没问题，但我不确定这对kernelsu和magisk有没有问题
> 
> 但这看起来并不是 APatch 的问题了，就像 bmax 所讲的那样，
> 
> > 模块安装的时候本来就是 root
> 
> 因为已经是最高权限了，su 是不存在的。如果有必要可以让模块作者改进。

https://github.com/j-hc/revanced-magisk-module/pull/409
这是一个都没问题的问题，祈祷能被解决🙏🙏🙏