bmax121/APatch #322: Module and root manager failed because of bug 错误导致的模块和超级用户管理器失效

pasical, 2024-03-24:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

系统启动一段时间后模块管理器就会失效，同时root授权页也会失效，在刚启动时可以正常运行。

### Reproduce method/复现方法

开机可能12小时左右

### Expected behavior/预期行为

-

### Actual behaviour /实际行为

如图

### Screenshots/截图

[image]
[image]

### Device Name/设备名称

xiaomi 13 pro

### OS Version/系统版本

hyperOS/Android14

### APatch Version/APatch 版本

10570

### Kernel Version/内核版本

-

### KernelPatch Version/KernelPatch 版本

-

### Additional context/其他信息

-

pomelohan, 2024-03-24:
请提供 bugreport，不然无法修复

pasical, 2024-03-24:
> 请提供 bugreport，不然无法修复

已上传

pomelohan, 2024-03-24:
> > 请提供 bugreport，不然无法修复
> 
> 已上传

都装了什么模块

pasical, 2024-03-24:
> > > 请提供 bugreport，不然无法修复
> > 
> > 
> > 已上传
> 
> 都装了什么模块

[image]
划红线的是没开的

pasical, 2024-03-25:
是否有可能和网络有关？🤔关闭AdguardHome后暂时未发生问题。

pomelohan, 2024-04-13:
请尝试最新 ci 版本是否仍然存在问题。

pasical, 2024-04-13:
目前是10657，刚才看的时候问题依然存在。我再观察几天
[Image]
这是目前的模块列表

pomelohan, 2024-04-13:
QingYue 关了看看

pasical, 2024-04-16:
目前是开着qingyue，没重启过，三天没问题

pasical, 2024-04-20:
更新：关闭qingyue后依然复现，猜测是否可能与爱玩机工具箱有关

pasical, 2024-04-28:
同样复现，近期进行过的操作：更新爱玩机工具箱、更新rex yt模块

pasical, 2024-05-10:
更新：在10763版本发布至今没有出现问题。

closed 2024-05-12 by pasical