bmax121/APatch #323: I patched boot.img and flashed it. Then when i boot and entered the correct Superkey the app will crash.

DAmniTza, 2024-03-24:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [ ] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

crashed when enter the correct Superkeyd

### Reproduce method/复现方法

s

### Expected behavior/预期行为

s

### Actual behaviour /实际行为

s

### Screenshots/截图

[image]
[image]
[image]
[image]

### Logs/日志

down

### Device Name/设备名称

vivo 1818 or vivo v15 pro

### OS Version/系统版本

lastest

### APatch Version/APatch 版本

lastest

### Kernel Version/内核版本

4.14.190-perf +

### KernelPatch Version/KernelPatch 版本

lastest

### Additional context/其他信息

[logcat.txt](https://github.com/bmax121/APatch/files/14735652/logcat.txt)

DAmniTza, 2024-03-24:
[APatch_bugreport_2024-03-24_17_35.tar.gz](https://github.com/bmax121/APatch/files/14735717/APatch_bugreport_2024-03-24_17_35.tar.gz)

DAmniTza, 2024-03-24:
garfield han says : It seems that it's a common issue on 4.14.xx kernel.

kpimg couldn't resolve tcp_init_sock 's location.

sekaiacg, 2024-03-27:
Try using kptools from 0.10.5 and fix manually.
https://github.com/bmax121/KernelPatch/releases/tag/0.10.5-dev
And upload your kernel.

pomelohan, 2024-03-30:
> garfield han says : It seems that it's a common issue on 4.14.xx kernel.
> 
> kpimg couldn't resolve tcp_init_sock 's location.

I'm sorry for the misunderstanding, as you didn't provide a screenshot of the patch at the beginning, but in reality, there is no problem with tcd_init_sock. These past few days, I have also encountered some vivo machines experiencing flashing after entering a password after updating a new system version. This seems to be due to the vivo kernel limiting root.

Skorpion96, 2024-05-01:
I don't know if this is relevant but vivo kernels have hardcoded a thing that will make it reject the su command, this is why solutions like https://github.com/4accccc/vivo-Magisk-Delta-suu and https://github.com/4accccc/vivo-Magisk-suu/ did born, maybe apatch can be adapted for this? It just needs to use another applet instead of su in theory, I may be wrong though