bmax121/APatch #325: 新UI方面的问题

149898400, 2024-03-24:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

新UI将卸载合并后依旧是只要一点击立刻就会把系统补丁卸载，感觉有待改进。另外将设置的入口从主页面右上角移到底栏，感觉底栏过于拥挤。

### Describe the solution you'd like/描述你想要的解决方案

希望点击卸载后不再是立刻卸载系统补丁然后进入新页面的方式，而是改为像原来的安装一样，弹出一个窗口，同样是三个选项，分别是：还原内核镜像、取消修补、卸载系统补丁，每个选项选择后都需要进行确认，避免误点击。
然后主页面点击设备那一栏弹出窗口信息（卸载界面的启动镜像的内容，a/b槽什么的）
点击内核那一栏弹出窗口信息（卸载界面的内核详细信息）
之后将右上角三个点的反馈和关于移到设置的最下面，并将设置的入口重新放回右上角，避免底栏太过拥挤

### Describe alternatives you've considered/描述您考虑过的备选方案

无

### Additional context/其他信息

无

pomelohan, 2024-03-25:
1. 该改动暂时未合并进入主线，在支线分支做开发，开发完毕后需要内部评估后再决定是否合并。因此，具体以 release 版本为主。
2. 设置的按钮位置暂时不考虑更换了，现在我们日用的很多软件都是有 5 个在下面（淘宝、支付宝、微信等）放到其他地方反而不协调
3. 还原内核镜像就是取消修补
4. `然后主页面点击设备那一栏弹出窗口信息（卸载界面的启动镜像的内容，a/b槽什么的）
点击内核那一栏弹出窗口信息（卸载界面的内核详细信息）` 这个在交互上有些多余，没有实际意义，暂不考虑
5. 卸载按钮可选卸载 kp / ap 这个建议很好，未来我们会实现，感谢你对本项目的支持和宝贵的建议🌹

pomelohan, 2024-03-25:
已经添加

closed 2024-03-25 by pomelohan