bmax121/APatch #340: root用户不能访问su

taherenfa, 2024-03-31:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

切换到root用户后，再次执行su会提示找不到su；为什么要这么做，是因为termux使用tmoe启动chroot容器时发现无法启动(找不到su)，而它就是类似的逻辑

### Reproduce method/复现方法

在root用户下执行su的一系列指令

### Expected behavior/预期行为

它理应会再次执行su相关的命令，例如su --help理应输出帮助菜单

### Actual behaviour /实际行为

找不到文件或文件夹，在Magisk & KernelSU上则正常执行了操作

### Screenshots/截图

[image]

### Device Name/设备名称

小米11Lite(renoir)

### OS Version/系统版本

Android13

### APatch Version/APatch 版本

发行版与最新CI均有尝试

### Kernel Version/内核版本

5.4.233

### KernelPatch Version/KernelPatch 版本

0.10.5(10622)

### Additional context/其他信息

[APatch_bugreport_2024-04-01_07_44.tar.gz](https://github.com/bmax121/APatch/files/14817970/APatch_bugreport_2024-04-01_07_44.tar.gz)

bmax121, 2024-04-01:
下一版修复

closed 2024-04-01 by taherenfa

bmax121, 2024-04-03:
现在的 ci 应该可用了

taherenfa, 2024-04-03:
> 现在的 ci 应该可用了

谢谢特意通知，昨天看到CI更新后就去测试过可以正常使用