bmax121/APatch #342: 密码错误

lxx12445, 2024-04-01:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

从工作流285开始的测试版，我输入正确的密码，无法获得授权

### Reproduce method/复现方法

下载最新测试版，输入密码，会显示错误

### Expected behavior/预期行为

无

### Actual behaviour /实际行为

密码正确，验证失败

### Device Name/设备名称

红米k40s

### OS Version/系统版本

miui14 a13

### APatch Version/APatch 版本

g5cb3e5f以上的版本

### Kernel Version/内核版本

4.19

### KernelPatch Version/KernelPatch 版本

0.10.5

### Additional context/其他信息

[Uploading APatch_bugreport_2024-04-02_07_40.tar.gz…]()

0x0021, 2024-04-02:
同样的问题，已退回 g5cb3e5f

bmax121, 2024-04-02:
0.10.5 还在开发中，后面会好，先别用

closed 2024-04-02 by bmax121