bmax121/APatch #344: System modules not validated

CMLNT, 2024-04-03:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

[image]
我安装了我自己创建的模块和Zygisk，自动重启后发现模块并未生效
检查/data/adb/modules/ 模块是update状态 
ap/下modules_update.img 存在
断电重启和 adb reboot 重启都不生效
我的机器使用magisk也需要使用特定的magisk mount list方法 

### Reproduce method/复现方法

- 1. app安装模块
- 2. 命令行安装模块都是如此

### Expected behavior/预期行为

模块正常安装

### Actual behaviour /实际行为

模块未能正常安装、且不能卸载

### Device Name/设备名称

B* Auto D*k 5

### OS Version/系统版本

12 api32

### APatch Version/APatch 版本

10630

### Kernel Version/内核版本

5.4.219-qgki

### KernelPatch Version/KernelPatch 版本

0.10.5

### Additional context/其他信息

[2.zip](https://github.com/bmax121/APatch/files/14844939/2.zip)

pomelohan, 2024-04-03:
“特定的magisk mount list方法” 指的是什么

CMLNT, 2024-04-03:
> “特定的magisk mount list方法” 指的是什么
请看这个提交
https://github.com/topjohnwu/Magisk/commit/9cfda542298ce6742c5f8278d4700a157543598e

pomelohan, 2024-04-03:
`Patch @ %08zX [ro.build.system.fission_single_os] -> [ro.build.system.xxxxxxxxxxxxxxxxx]`

我认为这种提交是不必要合并到主线 APatch 的，如果有需要可以自行修改来支持，如果 Magisk / KernelSU 官方版可以正常工作，APatch 不能工作，我们才会修复。

closed 2024-04-03 by pomelohan

CMLNT, 2024-04-03:
> `Patch @ %08zX [ro.build.system.fission_single_os] -> [ro.build.system.xxxxxxxxxxxxxxxxx]`
> 
> 我认为这种提交是不必要合并到主线 APatch 的，如果有需要可以自行修改来支持，如果 Magisk / KernelSU 官方版可以正常工作，APatch 不能工作，我们才会修复。

能大概说下我需要修改哪里吗 麻烦了

pomelohan, 2024-04-03:
请自行研究，我们无法提供帮助。

MIkeeJY, 2024-05-17:
@CMLNT 我在云机里遇到同样的问题了，有解决方案吗？

MIkeeJY, 2024-05-18:
> > “特定的magisk mount list方法” 指的是什么
> > 请看这个提交
> > [topjohnwu/Magisk@9cfda54](https://github.com/topjohnwu/Magisk/commit/9cfda542298ce6742c5f8278d4700a157543598e)

我的机器默认magisk 插件是可以的不需要patch

pomelohan, 2024-05-18:
> > > “特定的magisk mount list方法” 指的是什么
> > > 请看这个提交
> > > [topjohnwu/Magisk@9cfda54](https://github.com/topjohnwu/Magisk/commit/9cfda542298ce6742c5f8278d4700a157543598e)
> 
> 我的机器默认magisk 插件是可以的不需要patch

你的刷入官方 Magisk 可以用吗

MIkeeJY, 2024-05-18:
@pomelohan 刷入magisk 插件是可用的，Apatch 最新版 插件是灰色 不可点

MIkeeJY, 2024-05-20:
[image]
[image]
[APatch_bugreport_2024-05-20_10_13.tar.gz](https://github.com/bmax121/APatch/files/15371864/APatch_bugreport_2024-05-20_10_13.tar.gz)
最新版截图和log

MIkeeJY, 2024-05-20:
[image]
[image]
Root功能授权一切正常

pomelohan, 2024-05-20:
> ![image](https://private-user-images.githubusercontent.com/8580762/331924147-fa163ff8-cf76-4892-bb3f-b16a72bccd63.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]

我把你那个 issue 重新打开了，后面有空了再研究下什么情况。