bmax121/APatch #348: unable to patch boot and install modules with Zygisk Next on

Johce-Chi, 2024-04-03:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

the problem is that while ZygiskNext on patching boot or installing magisk moudles will cause Manager crash, switch off zygisk next those function can be used again.

### Reproduce method/复现方法

1.install ZygiskNext
2.reboot makes moudle take effect
3.try install other modules or patch boot,then Manager will crash and exit
4.switch off Zygisknext,reboot
5.try install other modules or patch boot,normal
[image]

### Expected behavior/预期行为

Install other modules or patch boot with ZygiskNext on

### Actual behaviour /实际行为

try install other modules or patch boot with ZygiskNext on,then Manager will crash and exit

### Logs/日志

[image]

### Device Name/设备名称

psyche/Xiaomi 12X

### OS Version/系统版本

HyperOS 1.0.24.3.18

### APatch Version/APatch 版本

10622

### Kernel Version/内核版本

4.19.157

### KernelPatch Version/KernelPatch 版本

0.10.5

### Additional context/其他信息

[APatch_bugreport_2024-04-03_08_59.tar.gz](https://github.com/bmax121/APatch/files/14846641/APatch_bugreport_2024-04-03_08_59.tar.gz)

pomelohan, 2024-04-03:
是否方便远程

Johce-Chi, 2024-04-03:
> 是否方便远程

明天可以的

pomelohan, 2024-04-04:
最新版 Zygisk Next + APatch CI 还是可以复现吗，如果可以的话可以试试卸载 APatch Manager 重新安装试试。

如果仍然有问题，请在触发一次后再到设置 -> 发送日志中重新发送一份，感谢🌹

Johce-Chi, 2024-04-04:
> 最新版 Zygisk Next + APatch CI 还是可以复现吗，如果可以的话可以试试卸载 APatch Manager 重新安装试试。
> 
> 如果仍然有问题，请在触发一次后再到设置 -> 发送日志中重新发送一份，感谢🌹

最新版Zygisk next和最新CI版Manager，问题依旧，同时出现问题的时候也获取不到日志文件，比如发送到微信提示资源获取失败，发送到QQ提示文件不存在，关了Zygisk再重启后的日志有效吗

pomelohan, 2024-04-04:
不行，如果这样的话最好还是远程，这是我联系方式:

t.me/pomelohan

closed 2024-04-04 by pomelohan

Johce-Chi, 2024-04-04:
有类似情况的的请将apatch从隐秘空间移除