bmax121/APatch #351: Rescuing bricks method

berlinwow, 2024-04-04:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

When certain modules installed and can enter the system normally after reboot, but the phone cannot be used normally, like black screen.
In this circumstance, the brick rescue function of Apatch will not be triggered. We need an additional triggering method, such as pressing the volume button.

### Describe the solution you'd like/描述你想要的解决方案

Add functionality to disable the module after pressing the volume button several times

### Describe alternatives you've considered/描述您考虑过的备选方案

No

bmax121, 2024-04-12:
Have you tried press the volume down several times when system bootimg

berlinwow, 2024-04-14:
Of course, I also tried using the save brick module and it didn't work properly either.

pomelohan, 2024-05-15:
@berlinwow Try latest release, we have fixed volume key pressing to enter safe mode

closed 2024-05-15 by pomelohan