bmax121/APatch #352: 功能请求： 请求将系统模块中支持webview的模块的”卸载“按钮与不支持webview的模块的”卸载"按钮对齐

M-ls, 2024-04-04:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

在浏览近期issues没有发现相关问题
Browsing recent issues, no relevant issues were found.

### Describe the solution you'd like/描述你想要的解决方案

希望将“卸载”按钮位置统一，将打开webview的按钮放到左边
I hope to unify the position of the "Uninstall" button and move the button to open the webview to the left

### Describe alternatives you've considered/描述您考虑过的备选方案

考虑将webview打开放到标题上并且存在该功能时标题不同颜色区分？
Consider opening the webview on the title and distinguishing the title with different colors when this function exists?

### Additional context/其他信息

[image]
在用户安装多个模块时，从上往下划的思维惯性使得上方的“卸载”按钮在下方模块存在多个按钮时潜意识认定右侧同位置的按钮为“卸载”功能，容易误触
When users install multiple modules, the mental inertia of swiping from top to bottom causes the "Uninstall" button at the top to subconsciously identify the button at the same position on the right as the "Uninstall" function when there are multiple buttons for the module below. It is easy to accidentally touch it.

pomelohan, 2024-04-04:
👍采纳，下个版本实现

pomelohan, 2024-04-05:
Finished on https://github.com/bmax121/APatch/commit/fecdfa5be19ef864483b14bca0fdb1433afaf5ec

closed 2024-04-05 by pomelohan