bmax121/APatch #359: system module conflict/多个模块修改system分区时，会导致特定模块失效

lzghzr, 2024-04-08:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

不太清楚标题描述是否准确，此bug仅在多个(本人测试为6个)模块同时对system分区修改时产生，产生bug时任意禁用一个模块bug消失

### Reproduce method/复现方法

安装超过6个对system分区进行修改的模块

### Expected behavior/预期行为

多个模块皆可生效

### Actual behaviour /实际行为

特定模块 (HideNavBar) 失效

### Device Name/设备名称

Sony XQ-DQ72

### OS Version/系统版本

14  (API 34)

### APatch Version/APatch 版本

10657

### Kernel Version/内核版本

5.15.94-android13-8-00003-g2ac3c8c88d7f-ab10835385

### KernelPatch Version/KernelPatch 版本

0.10.5

### Additional context/其他信息

[APatch_bugreport.zip](https://github.com/bmax121/APatch/files/14900245/APatch_bugreport.zip)

受影响模块为 [HideNavBar](https://github.com/Magisk-Modules-Alt-Repo/HideNavBar)

附件内包含产生bug时所启用的模块，由于25M体积限制，实际app已移除
`mount.ts` 为根据 `ksud` 所修改的源码，可以解决此bug，但本人对rust和overlay都不熟悉，所以不知道为何可以修复

Admirepowered, 2024-04-08:
https://github.com/bmax121/APatch/actions/runs/8604701850  test this version

lzghzr, 2024-04-08:
> https://github.com/bmax121/APatch/actions/runs/8604701850 test this version

[APatch_bugreport_2024-04-09_06_47.tar.gz](https://github.com/bmax121/APatch/files/14911593/APatch_bugreport_2024-04-09_06_47.tar.gz)
并未解决

Admirepowered, 2024-04-09:
Fixed

closed 2024-04-09 by Admirepowered