bmax121/APatch #364: Add an option to install modules or recovery zip

TDD788, 2024-04-09:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

There are several zips that are installed in Magisk but not in APatch

### Describe the solution you'd like/描述你想要的解决方案

Being able to install the zip of a recovery or the RO2RW or whatever zip 

### Describe alternatives you've considered/描述您考虑过的备选方案

I would like nothing more than compatibility that does not specifically need the module.prop to be relative

pomelohan, 2024-04-10:
I'm sorry but I couldn't understand your meaning.

Admirepowered, 2024-04-10:
upload your several zips which can't use

TDD788, 2024-04-11:
basically the zip installer of some custom recovery or in simple words an installer that does not need the module.prop to install
[Uploading RO2RW-TEST-3.7.3.0s.zip…]()

closed 2025-03-09 by Admirepowered