bmax121/APatch #368: Unable to use Sandbox in chroot

ask9027, 2024-04-10:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

unable to use sandbox in chroot while open chromium,
but if i root with magisk it will work, means it use sandbox in chroot. i can run chromium without `--no-sandbox` with magisk

### Reproduce method/复现方法

Install Linuxdeploy termux-x11 for gui (grant root permission from apatch)
install debian then chromium browser
open it. 
it will crash 
but if run with `--no-sandbox` flag it can run.

### Expected behavior/预期行为

chromium should run without `--no-sandbox` flag

### Actual behaviour /实际行为

through `Failed to connect to the bus: Failed to connect to socket /run/dbus/system_bus_socket: Connection refused`

### Device Name/设备名称

Mi 11 Lite 4G

### OS Version/系统版本

13

### APatch Version/APatch 版本

10657 github action

### Kernel Version/内核版本

4.14.190-perf-g019d15b3d168

### KernelPatch Version/KernelPatch 版本

0.10.6-dev

### Additional context/其他信息

chromium log from apatch
[apatch_chromium.txt](https://github.com/bmax121/APatch/files/14933323/apatch_chromium.txt)

from magisk
[magisk_chromium.txt](https://github.com/bmax121/APatch/files/14933325/magisk_chromium.txt)

pomelohan, 2024-04-19:
Did you try whether working on `SELinux Permissive` Status? I don't thought this is a bug that APatch should fix.

ask9027, 2024-04-19:
> Did you try whether working on `SELinux Permissive` Status? I don't thought this is a bug that APatch should fix.

i tried with both, it is working with magisk(or magisk delta), so ya, there is problem in APatch

ask9027, 2024-04-24:
also i check with KernelSu , it is working but still not working with APatch.

sherylynn, 2024-06-18:
same problem in chroot firefox

ask9027, 2024-09-15:
i got logs from chroot
when `--no-sandbox` flag enable
```bash
termux@localhost:~$ chromium --no-sandbox
[15346:15387:0915/102009.087223:ERROR:zygote_host_impl_linux.cc(272)] Failed to adjust OOM score of renderer with pid 15393: Permission denied (13)
[15346:15387:0915/102009.168564:ERROR:zygote_host_impl_linux.cc(272)] Failed to adjust OOM score of renderer with pid 15396: Permission denied (13)
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
[15393:15393:0915/102009.390195:ERROR:viz_main_impl.cc(166)] Exiting GPU process due to errors during initialization
[15346:15387:0915/102009.451098:ERROR:zygote_host_impl_linux.cc(272)] Failed to adjust OOM score of renderer with pid 15441: Permission denied (13)
[15346:15387:0915/102009.465684:ERROR:zygote_host_impl_linux.cc(272)] Failed to adjust OOM score of renderer with pid 15443: Permission denied (13)
[15346:15387:0915/102009.708358:ERROR:zygote_host_impl_linux.cc(272)] Failed to adjust OOM score of renderer with pid 15462: Permission denied (13)
[15346:15387:0915/102009.751182:ERROR:zygote_host_impl_linux.cc(272)] Failed to adjust OOM score of renderer with pid 15469: Permission denied (13)
[15346:15387:0915/102009.817348:ERROR:zygote_host_impl_linux.cc(272)] Failed to adjust OOM score of renderer with pid 15474: Permission denied (13)
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
[15469:15469:0915/102010.081265:ERROR:viz_main_impl.cc(166)] Exiting GPU process due to errors during initialization
[15346:15387:0915/102010.116773:ERROR:zygote_host_impl_linux.cc(272)] Failed to adjust OOM score of renderer with pid 15512: Permission denied (13)
[15346:15387:0915/102010.273517:ERROR:zygote_host_impl_linux.cc(272)] Failed to adjust OOM score of renderer with pid 15531: Permission denied (13)
[15346:15387:0915/102010.681164:ERROR:zygote_host_impl_linux.cc(272)] Failed to adjust OOM score of renderer with pid 15549: Permission denied (13)
[15346:15387:0915/102010.947408:ERROR:zygote_host_impl_linux.cc(272)] Failed to adjust OOM score of renderer with pid 15565: Permission denied (13)
[15346:15387:0915/102010.966129:ERROR:zygote_host_impl_linux.cc(272)] Failed to adjust OOM score of renderer with pid 15571: Permission denied (13)
[15346:15387:0915/102012.746285:ERROR:zygote_host_impl_linux.cc(272)] Failed to adjust OOM score of renderer with pid 15588: Permission denied (13)
```

and when without `--no-sandbox` means with `sandbox` env
```bash
termux@localhost:~$ chromium
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
[13661:13661:0915/101913.909939:ERROR:viz_main_impl.cc(166)] Exiting GPU process due to errors during initialization
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
[13726:13726:0915/101914.722007:ERROR:viz_main_impl.cc(166)] Exiting GPU process due to errors during initialization
error: XDG_RUNTIME_DIR is invalid or not set in the environment.
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
libunwind: __unw_add_dynamic_fde: bad fde: FDE is really a CIE
[13776:13776:0915/101915.077950:ERROR:viz_main_impl.cc(166)] Exiting GPU process due to errors during initialization
```

i hope this will help to find out issue why chromium not working in sandbox env while use APatch root method.

TreinTR, 2024-10-18:
I have the same problem in chroot, with both chromium and firefox.

closed 2025-03-09 by Admirepowered

ask9027, 2025-03-09:
I installed latest version and flash boot.img again
but still same problem that can't use sandbox.
apps are still crashed