bmax121/APatch #372: AndroidPatch not installed everytime app is opened

VibhavBilgoji, 2024-04-11:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Whenever I open APatch, it says that Androidpatch is not installed. When I press install, it does get installed, but again after I clear the app from the ram and reopen it, it shows the same thing.

### Reproduce method/复现方法

Open APatch app
Click on install (Androidpatch)
Clear the app from recents
Reopen the app

### Expected behavior/预期行为

It should be installed forever once I press install, or be installed by default after I flash the patched boot image

### Actual behaviour /实际行为

It automatically gets uninstalled everytime I close the app

### Screenshots/截图

[image]

### Logs/日志

[APatch_bugreport_2024-04-11_10_13.tar.gz](https://github.com/bmax121/APatch/files/14941547/APatch_bugreport_2024-04-11_10_13.tar.gz)

### Device Name/设备名称

Redmi Note 11

### OS Version/系统版本

14

### APatch Version/APatch 版本

10657

### Kernel Version/内核版本

4.19.304

### KernelPatch Version/KernelPatch 版本

0.10.5

### Additional context/其他信息

Even though the app says AndroidPatch is not installed, the installed modules work fine, so I wonder if this is just a UI glitch or the the Androidpatch actually gets uninstalled after I close the app

pomelohan, 2024-04-11:
Try to follow those steps to send a new bugreport:
1. Reboot your device
2. Open APatch, and now AndroidPatch shows that not installed
3. Record time
4. Install AndroidPatch
5. Re-open APatch, shows that AndroidPatch not installed
6. Record time

Send log here and time. Thank you.

VibhavBilgoji, 2024-04-11:
can i send you a screenrecord?
ill send google photos link, because github won't lemme upload >10 MB videos
https://photos.app.goo.gl/6FtKaZDQXwgXcpDv7

pomelohan, 2024-04-11:
Bugreport is better than screen record since I knew the behaviour.

VibhavBilgoji, 2024-04-11:
what do you exactly mean by "record time"?

pomelohan, 2024-04-11:
record time means clock, so that I can locate the bug fastly.

pomelohan, 2024-04-12:
https://github.com/bmax121/APatch/actions/runs/8663352621/artifacts/1408912921

Use this version and send bugreport again, follow the step as I mentioned. Thank you.

VibhavBilgoji, 2024-04-12:
Rebooted, installed Androidpatch patch at 9:21PM
 i.e. 21:21
[APatch_bugreport_2024-04-12_21_21.tar.gz](https://github.com/bmax121/APatch/files/14960846/APatch_bugreport_2024-04-12_21_21.tar.gz)

Cleared the app, reopened at 9:22PM, 21:22
Again Androidpatch not installed message 
[APatch_bugreport_2024-04-12_21_22.tar.gz](https://github.com/bmax121/APatch/files/14960854/APatch_bugreport_2024-04-12_21_22.tar.gz)

pomelohan, 2024-04-13:
> Rebooted, installed Androidpatch patch at 9:21PM i.e. 21:21 [APatch_bugreport_2024-04-12_21_21.tar.gz](https://github.com/bmax121/APatch/files/14960846/APatch_bugreport_2024-04-12_21_21.tar.gz) [...]

`ls -lh /data/adb && cd /data/adb && ./apd -V`

Run those commands on Terminal, show me the result by screenshots

VibhavBilgoji, 2024-04-13:
[image]

these are the modules i have already installed btw
[image]

pomelohan, 2024-04-13:
😂 How could modules work without AndroidPatch installed?

I need to send you a debug version of APK, but due to its size exceeding 20MB, it seems that it cannot be uploaded through GitHub attachment. If convenient, you can contact me through t.me/pomelohan

VibhavBilgoji, 2024-04-13:
"Even though the app says AndroidPatch is not installed, the installed modules work fine, so I wonder if this is just a UI glitch or the the Androidpatch actually gets uninstalled after I close the app"

i had already mentioned this in the og post

pomelohan, 2024-04-13:
Fixed on https://github.com/bmax121/APatch/commit/72f10ae577a25f5c506a9481b4083aebd3537dcc

commit 72f10ae5 (2024-04-13): manager: Use our cli's rootShell instead of defaultShell

Fix #372
closed 2024-04-13 by pomelohan