bmax121/APatch #374: samsung galaxy a 32 5g which a supported kernel request but the apatch happened error while patching the AP files.

cengzhijie3, 2024-04-11:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

My kernel support suits and supports the Samsung Galaxy a 32 5g is a supported kernel request, but it has an error while patching the AP files.

### Reproduce method/复现方法

My kernel support suits and supports the Samsung Galaxy a 32 5g is a supported kernel request, but it has an error while patching the AP files.

### Expected behavior/预期行为

My kernel support suits and supports the Samsung Galaxy a 32 5g is a supported kernel request, but it has an error while patching the AP files.

### Actual behaviour /实际行为

My kernel support suits and supports the Samsung Galaxy a 32 5g is a supported kernel request, but it has an error while patching the AP files.

### Screenshots/截图

[image]
[image]
[image]

### Logs/日志

[APatch_bugreport_2024-04-11_20_21.tar.gz](https://github.com/bmax121/APatch/files/14946070/APatch_bugreport_2024-04-11_20_21.tar.gz)

### Device Name/设备名称

SAMSUNG GALAXY A32 5G

### OS Version/系统版本

ANDROID13

### APatch Version/APatch 版本

10657

### Kernel Version/内核版本

4.14.186-25943992

### KernelPatch Version/KernelPatch 版本

A326BXXSACXC1_A326BOLMACXC1_SIN

### Additional context/其他信息

[APatch_bugreport_2024-04-11_20_21.tar.gz](https://github.com/bmax121/APatch/files/14946051/APatch_bugreport_2024-04-11_20_21.tar.gz)

pomelohan, 2024-04-14:
duplicated

#123 
#271

TDD788, 2024-06-23:
1. first unzip the AP that apatch is not magisk and then unzip the boot.img.lz4 and then patch the boot.img

closed 2025-03-09 by Admirepowered