bmax121/APatch #375: Change memory for module.img

Zenlua, 2024-04-11:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

Add storage like ksu to avoid the problem of not enough throughput if the flash module is heavy.

[image]

### Describe the solution you'd like/描述你想要的解决方案

As above

### Describe alternatives you've considered/描述您考虑过的备选方案

As above

### Additional context/其他信息

As above

Skorpion96, 2024-04-13:
This willl break TWRP backups, people still don't undesrstand this.....

xxjoe2, 2024-04-15:
agreed, this is why i switched from ksu to ap

Skorpion96, 2024-04-15:
> agreed, this is why i switched from ksu to ap

@Zenlua is trying (without knowing it I understand) to ruin apatch as well....

Skorpion96, 2024-04-15:
I really don't know how to pray APatch creator to close this as unfeasible..... Please don't ruin apatch as well.....

Zenlua, 2024-04-15:
If you don't want to increase the size of the img module file, then delete the img module because it's not suitable. If the module is faulty, leave it like magisk.

xxjoe2, 2024-04-15:
it really hard to handle that 1T  sparse file, i tried to do a backup under recovery,  but native twrp doesnt work (size exceeded). i backup with tar (sparse file supported) and it took more than hour to complete, just the adb folder!!

so pls dont go back to sparse file

0x0021, 2024-04-15:
我拒绝😡

Zenlua, 2024-04-16:
> it really hard to handle that 1T sparse file, i tried to do a backup under recovery, but native twrp doesnt work (size exceeded). i backup with tar (sparse file supported) and it took more than an hour to complete, just the adb folder!!
> 
> so pls dont go back to sparse file

Just compress module.img into zip and delete module.img it won't take up space.

closed 2024-04-16 by Admirepowered