bmax121/APatch #376: Bootloop on Nintendo Switch after flashing APatch

Skorpion96, 2024-04-12:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [ ] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

I patched boot from Lineage from switchroot and Crdroid from:https://github.com/Lost-Entrepreneur439/nvidia-tegra-x1-crdroid-releases/releases/tag/02-15-2024 but on both it was stucking on logo so i think APatch is incompatible with Nintendo Switch. All this said would be possible to make it compatible? I'm using a Nintendo Switch lite with a Modchip on latest 18.0.0 firmware on hekate and nyx. To flash the patched boot i used TWRP recovery. I will try now to compile KernelSU, let's see what will happen

### Reproduce method/复现方法

Extract boot image from lineage or crdroid or wathever rom zi, give it to APatch app, let it patch. After that reboot the nintendo switch to TWRP, then tap Install, install image, select the patched image, select the boot partition, then tap reboot. After doing all these steps the device will remain on switchroot lineage bootlogo

### Expected behavior/预期行为

Obliviously the expected behaviour is that it would pass to the necxt boot phase (bootanimation) and boot correctly

### Actual behaviour /实际行为

The device stucks on bootlogo after APatch image flash, while for example Magisk works, and it's not a problem of permisive kernel as that Crdroid rom i tried had it by default. 

### Logs/日志

About the logs maybe tomorrow i will try to do some, although since the device stucks on bootlogo maybe i won't be able to take them

### Device Name/设备名称

Nintendo Switch/Nintendo Switch Lite (my model)

### OS Version/系统版本

Android 11

### APatch Version/APatch 版本

Latest

### Kernel Version/内核版本

4.9.135

### KernelPatch Version/KernelPatch 版本

-4.9.135 ? i would say i lie i swear, but i gave you the sources so better than this....

### Additional context/其他信息

here are kernel sources:https://github.com/CTCaer/switch-l4t-kernel-4.9/tree/linux-dev
I just hope this won't be closed as would be great to have a good root on the switch and sadly magisk isn't anymore in my opinion as any app can see it. So please if possible let's fix APatch on the switch. Obliviously i will give a try on KernelSU if possible

pomelohan, 2024-04-19:
It's really hard to fix without device. If you have kernel source code, you can recompile kernel with `CONFIG_KALLSYMS=y and CONFIG_KALLSYMS_ALL=y` if possible.

Skorpion96, 2024-04-19:
Will give a try

MaiSan9999, 2024-06-08:
> Will give a try

hey did you manage to boot with apatch?
same issue flashed patched boot.img with trwp now it's stuck on bootlogo_android.bmp

Skorpion96, 2024-06-08:
> > Will give a try
> 
> hey did you manage to boot with apatch? same issue flashed patched boot.img with trwp now it's stuck on bootlogo_android.bmp

I still have to try, I think we have to build it or wait for apatch to support CONFIG_KALLSYMS_ALL=n

MaiSan9999, 2024-06-08:
> > > Will give a try
> > 
> > hey did you manage to boot with apatch? same issue flashed patched boot.img with trwp now it's stuck on bootlogo_android.bmp
> 
> I still have to try, I think we have to build it or wait for apatch to support CONFIG_KALLSYMS_ALL=n

before flashing the patched boot.img i checked for requirements and it actually output CONFIG_KALLSYMS_ALL=y so i don't know why it didn't work.
mariko oled

Skorpion96, 2024-06-08:
> before flashing the patched boot.img i checked for requirements and it actually output CONFIG_KALLSYMS_ALL=y so i don't know why it didn't work. mariko oled

Well.... The real problem here is that we can't do boot logs so knowing why it fails to boot is tricky

theAeon, 2024-07-15:
Same issue here w/ v1 unpatched switch and lineage 21 with a direct fastboot flash. Confirmed CONFIG_KALLSYMS=y, CONFIG_KALLSYMS_ALL=y, and CONFIG_DEBUG_KERNEL=y. Is there a way to convince it to expose the bootlogs?

MaiSan9999, 2024-07-15:
> Same issue here w/ v1 unpatched switch and lineage 21 with a direct fastboot flash. Confirmed CONFIG_KALLSYMS=y, CONFIG_KALLSYMS_ALL=y, and CONFIG_DEBUG_KERNEL=y. Is there a way to convince it to expose the bootlogs?

https://gbatemp.net/threads/how-to-make-a-joycon-rail-uart-to-capture-boot-log.625784/
only way i think.

theAeon, 2024-07-16:
Unfortunate. Honestly if I trusted myself to go anywhere near a soldering iron I'd do it but my hands are a bit shaky for that. Wonder if there's molded versions of these on aliexpress the way there are rcm jigs or something.

theAeon, 2024-09-13:
So, I was browsing my switch FS today and found out that apparently L4T dumps panics to a text file, too! Wild, that. Anyway.
[L4T_panic.txt](https://github.com/user-attachments/files/16999790/L4T_panic.txt)
I have a memory dump too if useful.