bmax121/APatch #386: 自动授权

lxx12445, 2024-04-18:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [ ] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

apatch会自动授权一部分新安装的软件

### Reproduce method/复现方法

无

### Expected behavior/预期行为

无

### Actual behaviour /实际行为

无

### Logs/日志

[Uploading APatch_bugreport_2024-04-18_10_17.tar.gz…]()

### Device Name/设备名称

k40s

### OS Version/系统版本

miui14 a13

### APatch Version/APatch 版本

0.10.6

### Kernel Version/内核版本

4.19

### KernelPatch Version/KernelPatch 版本

10721

### Additional context/其他信息

无

pomelohan, 2024-04-19:
已知问题，需要监听 UID 变化后对列表进行重构，内部已经完成了这个修复，后面会在优化实现后发布。

lxx12445, 2024-04-19:
ok

获取Outlook for Android<https://aka.ms/AAb9ysg>

pomelohan, 2024-04-19:
> ok 获取Outlook for Android<https://aka.ms/AAb9ysg>
> […](#)

临时的解决方案是在卸载/安装应用后，删除 /data/adb/ap/package_config，然后打开 manager 重新配置授权列表

djo5296, 2024-09-01:
related to #508 , but this issue is still open so i'll add here instead.

Facebook and SM Malls Online are also getting superuser access automatically. Where does one find /data/adb/ap/package_config ? is that per app? Any recommended app to do this on the phone

closed 2025-03-09 by Admirepowered