bmax121/APatch #387: 2 elements of UI lack system color theme

Macrohard-X, 2024-04-18:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

部分元件未使用系统主题配色

### Reproduce method/复现方法

进入安装界面，如截图所示

### Expected behavior/预期行为

旋转加载圈以及文本框描边使用系统颜色主题

### Actual behaviour /实际行为

颜色固定为紫色

### Screenshots/截图

[image]
[image]
[image]

### Device Name/设备名称

Xiaomi 13 Ultra

### OS Version/系统版本

HyperOS 1.0.6.0 / Android 14

### APatch Version/APatch 版本

10727 (10657-70-gd82e676)

### Kernel Version/内核版本

5.15.94-android13-8-00001-g171753a15b59-ab11001737

### KernelPatch Version/KernelPatch 版本

0.10.6

### Additional context/其他信息

无

pomelohan, 2024-04-19:
是从 APatch Manager 设置中改的颜色主题吗

Macrohard-X, 2024-04-19:
> 是从 APatch Manager 设置中改的颜色主题吗

不管设置成什么颜色主题，图中这两个元件都固定显示成`#711EE9`的颜色

pomelohan, 2024-04-19:
> > 是从 APatch Manager 设置中改的颜色主题吗
> 
> 不管设置成什么颜色主题，图中这两个元件都固定显示成`#711EE9`的颜色

我知道，我的问题是 是从管理器设置中选择的颜色主题还是说你换系统壁纸了，dynamicColor 上报的

Macrohard-X, 2024-04-19:
> > > 是从 APatch Manager 设置中改的颜色主题吗
> > 
> > 
> > 不管设置成什么颜色主题，图中这两个元件都固定显示成`#711EE9`的颜色
> 
> 我知道，我的问题是 是从管理器设置中选择的颜色主题还是说你换系统壁纸了，dynamicColor 上报的

从没换过系统壁纸

commit 91d4cfaa (2024-04-19): manager: Use material3 component for PatchesScreen

Fix #387
closed 2024-04-19 by pomelohan