bmax121/APatch #400: Infinite loading in superuser list

rj10rd, 2024-04-24:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Unable to load the installed app list in superuser; unable to install modules; just show black when installing a module; tried the latest stable (0.10.5) and CI build (0.10.6 10739); unable to patch another boot image.
[image]

### Reproduce method/复现方法

Just patch the boot image using apatch. (We can patch if we clear the data of the apatch app, but after that, the same issue persists.)

### Expected behavior/预期行为

Superuser list should load and modules should be installed normally.

### Actual behaviour /实际行为

[image]
[image]

### Device Name/设备名称

Poco X2

### OS Version/系统版本

Awaken os 4.4 (Android 14 qpr2)

### APatch Version/APatch 版本

0.10.6/0.10.5

### Kernel Version/内核版本

Linux version 4.14.340-openela-perf-g5a9fc1311c85 (nobody@android-build) (Android (10087095, +pgo, +bolt, +lto, -mlgo, based on r487747c) clang version 17.0.2 (https://android.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362), LLD 17.0.2) #2 SMP PREEMPT Tue Apr 23 14:47:05 IST 2024

### KernelPatch Version/KernelPatch 版本

Version 10739 (10657-82-gced622d)

### Additional context/其他信息

After clearing APatch data patching process:
[image]
[image]
[image]

pomelohan, 2024-04-30:
Does your kernel enable `CONFIG_KALLSYMS_ALL`?

rj10rd, 2024-05-02:
Here you go with kernel source(I don't have enough knowledge)- https://github.com/xiaomi-sm6150/android_kernel_xiaomi_sm6150

Xori71, 2024-05-30:
> Does your kernel enable `CONFIG_KALLSYMS_ALL`?

I faced this exact same issues with CONFIG_KALLSYMS_ALL=y.

GithubDns, 2024-06-04:
I can't patch boot img and iam have infinite loop in superuser please help me I can't uninstall the root 
[image]

fabioesantos, 2024-06-18:
I have the same issue an S10 in android 14 only, android 13 is fine.

I have CONFIG_KALLSYMS_ALL=y

ventureoo, 2024-07-31:
Same issue on POCO X3 NFC (surya) after upgrading from LineageOS 19.1 to LineageOS 21. Kernel has `CONFIG_KALLSYMS_ALL=y` enabled  in config: https://github.com/LineageOS/android_kernel_xiaomi_surya/blob/lineage-21/arch/arm64/configs/surya_defconfig#L39

karasudidntwakeup, 2024-08-06:
same here crdroid

Rachit-Pandey-2004, 2024-08-09:
Same here with poco m2 pro with lineage os andriod 14

egydoctor2010, 2024-08-10:
Same happen with me before stock rom android 13 support apatch well, it is running well after rebatch boot again

borisiua, 2024-08-19:
same issue with lineage os 21 (android 14) on POCO X3 NFC (surya). same infinite loading of su list, can't install apm modules - error message: `Error: No such file or directory (os error 2)`
few more details:
- su actually is installed. i can run it with `adb shell su`

Jonte-P, 2024-08-29:
I'm having the same issue with lineage os 21 on my OnePlus 7T. I am going to try another rom to see whether it works or not. 
I've tried the nightly release of Apatch to no avail.

ventureoo, 2024-10-22:
> Same issue on POCO X3 NFC (surya) after upgrading from LineageOS 19.1 to LineageOS 21. Kernel has `CONFIG_KALLSYMS_ALL=y` enabled in config: https://github.com/LineageOS/android_kernel_xiaomi_surya/blob/lineage-21/arch/arm64/configs/surya_defconfig#L39

I investigated this issue a bit more by running apatch in android studio and debugging via lolcat. Judging from these logs this issue is a duplicate of https://github.com/bmax121/APatch/issues/283

```
-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A  *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A  LineageOS Version: '21.0-20241014-NIGHTLY-surya'
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A  Build fingerprint: 'POCO/surya_eea/surya:12/RKQ1.211019.001/V14.0.5.0.SJGEUXM:user/release-keys'
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A  Revision: '0'
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A  ABI: 'arm64'
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A  Timestamp: 2024-10-22 18:09:25.723402002+0300
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A  Process uptime: 1s
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A  Cmdline: me.bmax.apatch:root:0
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A  pid: 8960, tid: 8960, name: main  >>> me.bmax.apatch:root:0 <<<
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A  uid: 0
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A  signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A  Abort message: 'No pending exception expected: java.lang.ClassNotFoundException: Didn't find class "com.topjohnwu.superuser.internal.RootServerMain" on path: DexPathList[[],nativeLibraryDirectories=[/system/lib64, /system_ext/lib64, /system/lib64, /system_ext/lib64]]
                                                                                                      at java.lang.Class dalvik.system.BaseDexClassLoader.findClass(java.lang.String) (BaseDexClassLoader.java:259)
                                                                                                      at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String, boolean) (ClassLoader.java:637)
                                                                                                      at java.lang.Class java.lang.ClassLoader.loadClass(java.lang.String) (ClassLoader.java:573)'
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A      x0  0000000000000000  x1  0000000000002300  x2  0000000000000006  x3  0000007fd6b65250
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A      x4  0000007703ec8000  x5  0000007703ec8000  x6  0000007703ec8000  x7  000000000004eec6
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A      x8  00000000000000f0  x9  00000076f2b2f390  x10 ffffff80fffffbdf  x11 0000000000000001
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A      x12 0000007fd6b63fd0  x13 0000000000000282  x14 0000007fd6b64080  x15 00001141edb26197
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A      x16 00000076f2bdafe0  x17 00000076f2bc6860  x18 000000770348c000  x19 0000000000002300
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A      x20 0000000000002300  x21 00000000ffffffff  x22 000000000000000b  x23 b400007538309dd9
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A      x24 000000000000000b  x25 0000007702d58ac0  x26 b40000757830ba50  x27 0000007fd6b655b0
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A      x28 0000007702d58ac0  x29 0000007fd6b652d0
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A      lr  00000076f2b66ad8  sp  0000007fd6b65230  pc  00000076f2b66b08  pst 0000000000000000
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A  12 total frames
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A  backtrace:
2024-10-22 18:09:25.820  8981-8981  DEBUG                   pid-8981                             A        #00 pc 000000000005ab08  /apex/com.android.runtime/lib64/bionic/libc.so (abort+168) (BuildId: 10f6580e623cb20b4044f6e6a4103b34)
2024-10-22 18:09:25.821  8981-8981  DEBUG                   pid-8981                             A        #01 pc 00000000008ed988  /apex/com.android.art/lib64/libart.so (art::Runtime::Abort(char const*)+600) (BuildId: e80ff12130cade1e1fbcdd24f28a8ca0)
[... cut ...]
2024-10-22 18:09:25.821  8981-8981  DEBUG                   pid-8981                             A        #11 pc 0000000000054748  /apex/com.android.runtime/lib64/bionic/libc.so (__libc_init+116) (BuildId: 10f6580e623cb20b4044f6e6a4103b34)
2024-10-22 18:09:25.832  1503-2231  NativeCrashListener     system_server                        W  Couldn't find ProcessRecord for pid 8960
2024-10-22 18:09:25.835   680-680   tombstoned              tombstoned                           E  Tombstone written to: tombstone_11
2024-10-22 18:09:25.838  1503-2067  NativeTombstoneManager  system_server                        E  Tombstone's UID (0) not an app, ignoring
```

ventureoo, 2024-10-22:
I did a rebase of the patches from @pomelohan that he suggested here https://github.com/bmax121/APatch/issues/283#issuecomment-1997667190 and they fix the issue for me, so will share the latest version for everyone: https://github.com/ventureoo/APatch/tree/fix-infinite-loading

APK for testing: https://drive.google.com/file/d/1ShC4IdoLlnK7r6WAxtmTT2VPuDsJ95VB/view (please do NOT report any further issues with this APK as it is not official or signed).

rj10rd, 2025-03-05:
> I did a rebase of the patches from [@pomelohan](https://github.com/pomelohan) that he suggested here [#283 (comment)](https://github.com/bmax121/APatch/issues/283#issuecomment-1997667190) and they fix this issue for me, so will share the latest version for everyone who has had this issue: https://github.com/ventureoo/APatch/tree/fix-infinite-loading
> 
> APK for testing: https://drive.google.com/file/d/1ShC4IdoLlnK7r6WAxtmTT2VPuDsJ95VB/view (please do NOT report any further issues with this APK as it is not official or signed).

Su list works fine but unable to install any modules.

rj10rd, 2025-03-05:
In the test apk su list works fine but unable to install any modules.

midozalouk, 2025-04-16:
i think this is the fix , 

[](https://github.com/sidex15/android_kernel_lge_sm8150/commit/eaa448ec761528f7ccdf3ae998b8d24ba95a8e35l)

but needs to be implemented in Apatch

midozalouk, 2025-04-17:
> I did a rebase of the patches from [@pomelohan](https://github.com/pomelohan) that he suggested here [#283 (comment)](https://github.com/bmax121/APatch/issues/283#issuecomment-1997667190) and they [...]

Can you give me a guide how you edited the app and I will try to edit it to Make it work 

Thanks in advance