bmax121/APatch #408: Rescue feature Key combo for Apatch Module or Kernel Patch Module deactivation

Criticise, 2024-04-29:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

No

### Describe the solution you'd like/描述你想要的解决方案

Please provide a solution for Modules that lock the phone in a boot loop, the traditional Magisk Rescue deactivation modules no longer work on Apatch 

### Describe alternatives you've considered/描述您考虑过的备选方案

Magisk module for recovery mode
自动神仙救砖-支持OTA稳定版v35
音量键极速救砖
All not working

Criticise, 2024-04-30:
And un-installation feature or clean installation options for apatch, thank you

Skorpion96, 2024-05-01:
Not to make self advertising but I have modified the magisk manager for recovery mode to support kernelsu and apatch, if you look at my repos there is... People should "search on Google" the module is made to be executed in recovery terminal or adb shell on recovery and can disable modules or uninstall them...

Criticise, 2024-05-04:
> Not to make self advertising but I have modified the magisk manager for recovery mode to support kernelsu and apatch, if you look at my repos there is... People should "search on Google" (I don't want to piss but...) the module is made to be executed in recovery terminal or adb shell on recovery and can disable modules or uninstall them...

Thank you, any kind of help is appreciated 
Thank you always!

Skorpion96, 2024-05-04:
> Thank you, any kind of help is appreciated Thank you always!

Well in case here is the repo with the "mod" I had made, I don't know if it will be useful since seems that apatch will now get those functions from what I understood:https://github.com/Skorpion96/KernelSU-APatch-Manager-for-Recovery-Mode

Darlin19902015, 2024-10-20:
Alguien puede ayudarme tengo Apatch todo normal pero al actualizar a su última versión tengo un error al momento de parchar el boot del sistema alguien que me ayude solo sale error y un poco de letras 

https://github.com/user-attachments/assets/089a9ce0-458b-440a-a821-4ec1fb37cd19

closed 2025-03-09 by Admirepowered