bmax121/APatch #410: Regarding the issue of OPPO, OnePlus, and Realme series phones getting stuck during runtime when using APatch

yuchenytm, 2024-04-29:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Hello APatch developer, I am a Chinese user and my English is not good. Therefore, all descriptions in English are machine translated. Please understand that I will provide both Chinese and English descriptions. You can check the machine translation in English, translate my description in Chinese yourself, or request someone proficient in Chinese to translate it.

During my usage, I found that ColorOS13, ColorOS14, OxygenOS13, OxygenOS14 (and earlier versions of these systems may have this issue) can successfully obtain root after using APatch to patch boot partitions and flush them in. However, there is a possibility that the password input interface or the second Android loading interface after startup may get stuck, and after waiting for a while, it will automatically restart. Sometimes, when booting up, everything is normal, and all APatch functions can also be used normally. I hope you can identify the cause of this problem and fix it

I personally used the problematic device：
OnePlus11, System OxygenOS13
RealmeGT5 (Chinese version), system ColorOS14

APatch的开发者你们好，我是一位中国用户，我的英文不好，因此描述中的英文均为机器翻译，请你们谅解，我会提供中文和英文两个描述，你们可以查看机器翻译的英文，或者自行翻译我描述的中文，或请求熟练中文的人进行翻译。

在我的使用过程中，我发现，ColorOS13，ColorOS14，OxygenOS13，OxygenOS14（以及这些系统的更早版本都有可能出现这样的问题）在使用APatch修补boot分区并刷入后，可以成功获取root，但有概率会在启动后输入密码界面或者启动后的第二屏安卓加载界面卡住，等待一会后会自动重启，有的时候开机则一切正常，所有APatch的功能也均能正常使用，希望你们能够查明出现这个问题的原因并修复。

本人实测出现问题的设备：
OnePlus11，系统OxygenOS13
RealmeGT5（中国版），系统ColorOS14

### Reproduce method/复现方法

After using APatch to patch the boot, the problem described in the description can appear. I have preliminarily ruled out other reasons and roughly determined that it is some kind of problem with APatch

刷入APatch修补后的boot即可出现描述中的问题，本人已初步排除其他原因，大概确定是APatch出现的某种问题

### Expected behavior/预期行为

normal operation

正常运行

### Actual behaviour /实际行为

System crash occurred

出现系统崩溃

### Device Name/设备名称

OnePlus11

### OS Version/系统版本

OxygenOS13.0.1.596，Android13

### APatch Version/APatch 版本

10570

### Kernel Version/内核版本

5.15.74

### KernelPatch Version/KernelPatch 版本

0.10.4

### Additional context/其他信息

not have

无

pomelohan, 2024-04-30:
直接用中文就可以，目前活跃的开发者都是国人开发者。

`卡住，等待一会后会自动重启` 其实是由于 hook 过程中没有正确地处理 qit 跳转导致了 Kernel Panic，是已知问题，在 5.10+ 内核上可以复现，目前正在想办法修复，请等待后续版本。

BackTrackCRoot, 2024-05-20:
4.x pixel 3也出现bootloop 回退上一个版本OK

devnoname120, 2024-06-08:
Same problem on Mi Note 10 Lite except that it ways reboots, even if I'm fast enough when I type my unlock code and the home screen appears it reboots.

```shell
~ $ neofetch
         -o          o-            u0_a320@localhost
          +hydNNNNdyh+             -----------------
        +mMMMMMMMMMMMMm+           OS: Android 14 aarch64
      `dMMm:NMMMMMMN:mMMd`         Host: Xiaomi Mi Note 10 Li
      hMMMMMMMMMMMMMMMMMMh         Kernel: 4.14.345-openela-V
  ..  yyyyyyyyyyyyyyyyyyyy  ..     Uptime: 2 hours, 12 mins
.mMMm`MMMMMMMMMMMMMMMMMMMM`mMMm.   Packages: 80 (dpkg), 1 (pk
:MMMM-MMMMMMMMMMMMMMMMMMMM-MMMM:   Shell: bash 5.2.26
:MMMM-MMMMMMMMMMMMMMMMMMMM-MMMM:   CPU: Qualcomm SDMMAGPIE (8
:MMMM-MMMMMMMMMMMMMMMMMMMM-MMMM:   Memory: 3650MiB / 5506MiB
:MMMM-MMMMMMMMMMMMMMMMMMMM-MMMM:
-MMMM-MMMMMMMMMMMMMMMMMMMM-MMMM-
 +yy+ MMMMMMMMMMMMMMMMMMMM +yy+
      mMMMMMMMMMMMMMMMMMMm
      `/++MMMMh++hMMMM++/`
          MMMMo  oMMMM
          MMMMo  oMMMM
          oNMm-  -mMNs
```

Kernel source: https://github.com/PixelOS-Devices/kernel_xiaomi_sm6150

devnoname120, 2024-06-08:
I opened an issue because I'm not sure it's the same problem: https://github.com/bmax121/APatch/issues/495

closed 2025-03-09 by Admirepowered