bmax121/APatch #416: BankID app doesn't work

asamofal, 2024-05-01:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Hi there! I have a problem with the [BankID app](https://play.google.com/store/apps/details?id=no.vipps.bankid). There's a crash when I try to launch it. After removing AndroidPatch, the app starts working again.

### Reproduce method/复现方法

1. Flash patched boot image.
2. Launch the app.
3. See the carsh.

### Expected behavior/预期行为

[image]

### Actual behaviour /实际行为

[image]
[image]
[image]

### Device Name/设备名称

Google Pixel 7

### OS Version/系统版本

Android 14 (AP1A.240305.019.A1)

### APatch Version/APatch 版本

10657

### Kernel Version/内核版本

14 (API 34)

### KernelPatch Version/KernelPatch 版本

0.10.5

### Additional context/其他信息

[APatch_bugreport_2024-05-01_14_39.tar.gz](https://github.com/bmax121/APatch/files/15177091/APatch_bugreport_2024-05-01_14_39.tar.gz)

asamofal, 2024-05-02:
Update to 10763 didn't help me.

[image]

pomelohan, 2024-05-15:
Did you try to install Zygisk Next and add BankID to exclude list, and make Zygisk Next Enforce Denylist?

asamofal, 2024-05-15:
@pomelohan Zygisk Next was installed but I didn't know about the "Enforce Denylist" option there, enabling it helped! Many thanks to you.

Just to make it clear for anyone who come to this topic:
1. On `Superuser` tab choose target app and enable "Exclude Modifications".
2. Install Zygisk Next module. In the module settings enable "Enforce Denylist" option.

asamofal, 2025-02-13:
Release [11039](https://github.com/bmax121/APatch/releases/tag/11039) fixing the problem 🙂

closed 2025-02-13 by asamofal