bmax121/APatch #423: APatch 10763 Samsung S23u bootloop

xxmcvapourxx, 2024-05-02:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

 APatch 10763 patched kernel 0.10.7 bootloops. going back to 10570 0.10.5 boots fine.  

### Reproduce method/复现方法

Installed 10763 apatch app - direct patched to 0.10.7 and rebooted - S23u warning logo pressed power and reboots.
Went into twrp - choose zip option which switched to img - selected patched apatch 10763 - choose boot - wipe cache - rebooted. s23u warning pressed power to continue and bootloops.

went back to twrp flashed previous kernel 0.10.5 and wipe cache - rebooted - s23u warning logo - pressed power - boots fine. 

### Expected behavior/预期行为

After patching kernel directly in the latest kernel it patched fine and rebooted, i expected it to boot fine like previous kernels built. 

i had to go back to old version kernel that worked and not update the app. 

### Actual behaviour /实际行为

bootloop after patching s23u kernel with 10763 app. 

### Screenshots/截图

[image]
[image]
[image]
[image]
[image]

### Logs/日志

[last_kmsg.txt](https://github.com/bmax121/APatch/files/15184802/last_kmsg.txt)

### Device Name/设备名称

Samsung S23u SM-S918B/DS

### OS Version/系统版本

6.1 Android 14 S918BXXU3CXCF

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

6.1.78

### KernelPatch Version/KernelPatch 版本

0.10.5 > 0.10.7 Bootloop

### Additional context/其他信息

[last_kmsg.txt](https://github.com/bmax121/APatch/files/15184863/last_kmsg.txt)

crystle01, 2024-05-03:
I experience the same bug on another device too. Boot image with 0.10.7 causes device not to boot into Android. Reverting back to 0.10.5 works fine.

pomelohan, 2024-05-04:
> I experience the same bug on another device too. Boot image with 0.10.7 causes device not to boot into Android. Reverting back to 0.10.5 works fine.

Can you share your last_kmsg here?

I should confirm the bug

crystle01, 2024-05-04:
> > I experience the same bug on another device too. Boot image with 0.10.7 causes device not to boot into Android. Reverting back to 0.10.5 works fine.
> 
> Can you share your last_kmsg here?
> 
> I should confirm the bug

Yeah sure, here it is.
[last_kmsg.log](https://github.com/bmax121/APatch/files/15209603/last_kmsg.log)

crystle01, 2024-05-05:
> Yeah sure, here it is. [last_kmsg.log](https://github.com/bmax121/APatch/files/15209603/last_kmsg.log)

After entering pin for decryption device completely freezes and need to force reboot, unable to access device via adb after this point. Only via recovery. Were you able to see the replication in my log?

pomelohan, 2024-05-05:
Checked but no panic log, it shows it reboot with power key pressed. I am busy in next few days.

crystle01, 2024-05-06:
> Checked but no panic log, it shows it reboot with power key pressed. I am busy in next few days.

This one should have panic log, thanks for your time.
[last_kmsg.log](https://github.com/bmax121/APatch/files/15215582/last_kmsg.log)

sevsman, 2024-05-12:
I have the same issue on the S23 Ultra. xxmcvapourxx and I are in the same TG group.

xxmcvapourxx, 2024-05-18:
[last_kmsg.txt](https://github.com/bmax121/APatch/files/15365220/last_kmsg.txt)

updated to 10784 apatch version.

tried again 0.10.5 > 0.10.7 but getting bootloop see last ksmsg error.

xxmcvapourxx, 2024-05-26:
[APatch_bugreport_2024-05-26_10_29.tar.gz.zip](https://github.com/bmax121/APatch/files/15446954/APatch_bugreport_2024-05-26_10_29.tar.gz.zip)

just updated my phone to latest firmware and tried to patch kernel again. and i tried to save logs but its 0kbs.

still using 0.10.5. with Apatch version 10799 still boot loop on 0.10.7 when patching kernel directly @pomelohan

crystle01, 2024-05-26:
> Checked but no panic log, it shows it reboot with power key pressed. I am busy in next few days.

Hey man, did you have time to have a look? I haven't got into kernels or the works of this project so I'm unable to see..

xxmcvapourxx, 2024-05-26:
> > Checked but no panic log, it shows it reboot with power key pressed. I am busy in next few days.
> 
> Hey man, did you have time to have a look? I haven't got into kernels or the works of this project so I'm unable to see..

Dude please create your own thread. Don't use mine. This is for s23u. Use your own investigation like mine.

pomelohan, 2024-06-04:
> > Checked but no panic log, it shows it reboot with power key pressed. I am busy in next few days.
> 
> Hey man, did you have time to have a look? I haven't got into kernels or the works of this project so I'm unable to see..

Most of this project’s maintainers are busy in life, it may still need some times.

pomelohan, 2024-06-15:
Hello everyone try latest ci:
https://github.com/bmax121/APatch/actions/runs/9526790301/artifacts/1604570412

xxmcvapourxx, 2024-06-15:
> Hello everyone try latest ci: https://github.com/bmax121/APatch/actions/runs/9526790301/artifacts/1604570412

I've tested this with another kernel it boots fine but when going into the phone "swipe to get into phone" it does not allow it. 

i know kernel is fine cause i booted it without patching a patch.

[image]
[image]
[image]
[image]
[image]

i could not get logs for you, but i did notice an error on patching.

xxmcvapourxx, 2024-06-19:
> Hello everyone try latest ci: https://github.com/bmax121/APatch/actions/runs/9526790301/artifacts/1604570412

I've tested going from  0.10.5 to 0.11.0-dev, boots fine but still unable to get into my phone. so went back to 0.10.5 same kernel and i can get into my phone,

xxmcvapourxx, 2024-06-28:
> Hello everyone try latest ci: https://github.com/bmax121/APatch/actions/runs/9526790301/artifacts/1604570412

[dmesg.log](https://github.com/user-attachments/files/16036688/dmesg.log)

finally managed to get a dmsg for you... i've tried permissive and enforced on that 0.11.0 dev. and lock screen and finger print removed same issue. phone doesnt move anything

closed 2025-03-09 by Admirepowered