bmax121/APatch #427: **Color Theme** interface abnormality

armv7a, 2024-05-04:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Clicking on Color Theme interface to switch colors results in abnormal interface.

### Reproduce method/复现方法

1. Click "Settings".
2. Uncheck "System Color Theme".
3. Click "Theme Color".
4. Click on any color.

### Expected behavior/预期行为

It will not cause any abnormal interface.

### Actual behaviour /实际行为

The theme color is switched correctly, but the interface for changing colors appears abnormal.

### Screenshots/截图

https://github.com/bmax121/APatch/assets/147451408/b7d70f2f-6cf1-4be4-a60f-b67f2387702a

### Device Name/设备名称

Redmi Note 13 Pro+

### OS Version/系统版本

MIUI 14.0.1

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

5.15.78

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

Nothing

pomelohan, 2024-05-05:
Do you mean it takes a long time before colorThemePicking-dialog disappeared?

armv7a, 2024-05-05:
> Do you mean it takes a long time before colorThemePicking-dialog disappeared?

yes

pomelohan, 2024-05-05:
> > Do you mean it takes a long time before colorThemePicking-dialog disappeared?
> 
> yes

But I can't reproduce it on my device and I have no idea about this.

pomelohan, 2024-05-08:
Send logcat here after reproducing the bug, I thought it was caused by Compose side.

pomelohan, 2024-05-15:
@armv7a Can you try this build?
https://github.com/bmax121/APatch/actions/runs/9086289036/artifacts/1502811330

pomelohan, 2024-05-23:
Done in 2d852e7

closed 2024-05-23 by pomelohan