bmax121/APatch #439: adb shell使用su命令提示  /system/bin/su: No such file or directory

jecelyin, 2024-05-09:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

adb shell 之后，死活拿不到root，其他应用可以正常root
```bash
lime:/ $ su
/system/bin/sh: /system/bin/su: No such file or directory
```

### Reproduce method/复现方法

已经试过unchecked com.android.shell 重启系统后再checked，然后adb shell 还是不行

### Expected behavior/预期行为

如何做可以正常su，ls查看  /system/bin/su 是存在的

### Actual behaviour /实际行为

 /system/bin/sh: /system/bin/su: No such file or directory

### Device Name/设备名称

Redmi Note 9 4G

### OS Version/系统版本

12

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

4.19.157

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

boot.img是通过Manager应用右上角Patch得到新的boot.img然后fastboot flash boot上去

jecelyin, 2024-05-09:
logs
[Uploading APatch_bugreport_2024-05-09_14_12.tar.gz…]()

jecelyin, 2024-05-09:
[APatch_bugreport_2024-05-09_14_12.tar.gz](https://github.com/bmax121/APatch/files/15260697/APatch_bugreport_2024-05-09_14_12.tar.gz)

jecelyin, 2024-05-09:
我把 termux 的自带的 usr 目录copy到 /data/local/tmp/下面，然后使用termux的bash命令，再调用 su 就可以了，估计是系统自带的bash少了点啥

rayshabh, 2024-05-12:
> boot.img是通过Manager应用右上角Patch得到新的boot.img然后fastboot flash boot上去

I'm facing the same issue. Below are the screenshots while performing `su` command on both Windows Terminal and Termux Terminal.
_Note:_ `com.android.shell` _is given superuser privileges via APatch!_

**On Windows:**
[image]

**On Termux:**
[image]

What I know about APatch is that it uses good features of both KernelSU and Magisk. I also know that KernelSU tricks the shell into thinking that `/system/bin/su` exists on the device when it is not. KernelSU does allow `su` to work in `adb shell`, but APatch cannot.

**APatch Version:**
[image]
[image]

Ai686Leo, 2024-05-21:
可以使用爱玩机之类的软件让这类软件强制被自己调用root几次就好了[image]

jkriyabd, 2024-06-17:
still now, i am facing problem in nethunter app. not gained root access at all.

ascorbic-acid, 2024-08-22:
same issue for me, when i give root access it give no such directory

PsyNyde, 2024-09-08:
i have the same issue. I've checked `/system/bin` directory but there's no `su` binary there. i tried `/storage/emulated/0/su` as `su` path but still it doesn't generate. if i name the `su` binary to already existing directory it works. (for example `/system/bin` removes the `/system/bin/` folder and replaces with a `bin` binary)

```sh
[18360.176997] [+] KP I su_reset_path: /system/bin/su
[18360.403219] IRQ6 no longer affine to CPU4
[18361.420778] type=1400 audit(1725793363.022:309): avc:  denied  { search } for  comm="libkpatch.so" name="tests" dev="mmcblk0p89" ino=1474564 scontext=u:r:untrusted_app:s0:c190,c256,c512,c768 tcontext=u:object_r:shell_test_data_file:s0 tclass=dir permissive=0 app=me.bmax.apatch
[18361.421231] type=1400 audit(1725793363.022:310): avc:  denied  { search } for  comm="libkpatch.so" name="tests" dev="mmcblk0p89" ino=1474564 scontext=u:r:untrusted_app:s0:c190,c256,c512,c768 tcontext=u:object_r:shell_test_data_file:s0 tclass=dir permissive=0 app=me.bmax.apatch
[18361.421330] type=1400 audit(1725793363.022:311): avc:  denied  { search } for  comm="libkpatch.so" name="tests" dev="mmcblk0p89" ino=1474564 scontext=u:r:untrusted_app:s0:c190,c256,c512,c768 tcontext=u:object_r:shell_test_data_file:s0 tclass=dir permissive=0 app=me.bmax.apatch
[18361.421413] type=1400 audit(1725793363.022:312): avc:  denied  { search } for  comm="libkpatch.so" name="tests" dev="mmcblk0p89" ino=1474564 scontext=u:r:untrusted_app:s0:c190,c256,c512,c768 tcontext=u:object_r:shell_test_data_file:s0 tclass=dir permissive=0 app=me.bmax.apatch
[18361.436995] [+] KP I commit_su: pid: 31197, tgid: 972, to_uid: 0, sctx: u:r:magisk:s0, via_hook: 0
```
i've found this in dmesg output which seems to be causing the error.

tabs, 2024-10-23:
一样的问题

hg42, 2024-11-25:
FYI:
* I am new to APatch (I am used to KSU for some time)
* APatch 0.10.7 10763
* test device is a Xiaomi Redmi Note 13 NFC (sapphiren)
* I am contributor to the NeoBackup project (hg42x on Telegram)

I (first) patched the boot image in my Ofox backup with APatch app and flashed it with fastboot.
After reboot, APatch said it's ok.

If I entered the **Superuser tab**, I got a **spinner**, I waited a long time, but it did not finish.

As I am impatient (sometimes), I tried adb in the meantime.
**In adb I got the described problem**.

with kp configured (this was the default)
in adb:
* `which kp` shows `/system/bin/kp`
* running or accessing `kp` says: `No such file or directory`

I noticed that KMP was not installed, so I thought, that could eventually be a reason (despite nobody says something similar, as far as I read).

So I patched boot again, but this time directly in APatch app.
* now it works (tested in NeoBackup, another story, see below),
* for unknown reason (I didn't investigate further) I have no KMP installed (I had it in between),
* the spinner worked in between, but currently I get the infinite spinner again (no clue what changed, I just restored my whole system, with NeoBackup), if I install KMP again (via button, then kill and restart APatch) the spinner works (well gathering the list, to be more exact)

as a conclusion from this alone:
* the spinner does not matter for a working su
* KMP does not matter for a working su
* **reinstalling the patch seemed to do the job**

* the problem reappeared...
    * as I said KMP was missing and the spinner didn't finish
    * I installed KMP and the spinner works
    * in adb
        * `$ echo $(which su kp)` -> `/system/bin/su`su
        * `$ su` -> `... No such file or directory`
        * `$ echo echo test | $(which su kp)` -> `test`
        * so it works with piping commands

**NeoBackup vs KSU vs vs su/kp**

Some time ago we had several users that tried to use NeoBackup with **KSU** (!) and other unusual root solutions.
KSU changed in between and replaced it's minimal su with sh, which solved the problem on their side.

For such situations, I recently changed the whole behavior of NeoBackup to use piping commands into a suCommand that is configurable (to really work, you need my test builds, called pumpkins, from telegram group).
NeoBackup tries suCommand first and then a bunch of fixed commands and takes the first that fulfills a condition (access to certain directories).
The suCommand is usually configured as
  `su -c 'nsenter --mount=/proc/1/ns/mnt sh'`

this does two things, become root and enter the mount namespace of init process.
(Note, root alone does not give access to all directories automatically.)

Letely ee had several users that tried to use NeoBackup with **APatch**.
I did not really understand why the improved scheme did not work with APatch.
I also got a new device now and try to use APatch myself.

So, I investigate how APatch works.

I see, that kp can not take a command line.
It also doesn't work like a shell.
So the standard suCommand does not work.
It took some time to find a suCommand that works with kp.
It has to pipe the commands into kp, then the stdin for the commands (a.g. input satream to tar).
The construct is complicated and needs the help of a sub-shell. For this the suCommand itself needs a shell, but kp isn't a shell.
Additionally, a few commands cannot use libsu because it cannot handle input and output streams.
The standard java functions use the first word as the command.
I ended up with this complicated contruct:
`sh -c "(echo 'nsenter --mount=/proc/1/ns/mnt sh'; cat) | su`
or if you are using kp
`sh -c "(echo 'nsenter --mount=/proc/1/ns/mnt sh'; cat) | kp`

For quick commands this is quite slow.
Fortunately most commands are done via libsu and use the single shell that libsu manages as a shell server.
It is still noticable, that backups of small apps are slower.

Also, I am quite sure there are a lot of apps that also use (some) su commands with parameters.
adb is often used like this: `adb shell su -c ls /data/data`

Even before KSU, su implementations varied according to the way they interpret the command line (poor design about quoting, escaping etc.). So more complicated command lines are usually piped into su.
A former method was to pipe commands into `su 0`, which is a normal parameter for a real su. It seems nobody uses this any more (but old apps might do this).

**Suggestion:**

you should probably go the same path like KSU did before (I actually assumed you would have already taken that route, because you claimed taking the best of Magisk and KSU :-) no offense).

The KSU solution is mentioned here:
https://github.com/XayahSuSuSu/Android-DataBackup/issues/234#issuecomment-2116961763
which links to https://github.com/tiann/KernelSU/tree/main/userspace/su

Perhaps you could add a setting to choose which shell to use, so we could also set a real su command if necessary (e.g. for some older apps, that don't get updates any more).

hg42, 2024-11-26:
with this new knowledge NeoBackup can now use:

`sh -c "exec su -c 'nsenter --mount=/proc/1/ns/mnt sh'"`

but still three sh or su to start

hg42, 2024-11-26:
It seems weird that my connectbot root@localhost config is configured with `exec su`.
Either this was a default at that time, or I entered it myself.
It maybe that I had this problem in the past (long ago) with another su...
maybe an old Magisk or even SuperSU.?

Is it possible, that your su code is from old origins?

hg42, 2024-11-27:
I installed **APatch 11007** from APatch_ci telegram group.

With this version **almost all problems are solved**.

NeoBackup can use it's normal procedure, using the default:
su -c 'nsenter --mount=/proc/1/ns/mnt sh'

I can use `su` to switch to superuser mode in Termux and other terminal apps.

in adb:

`su` works

`file /system/bin/su` still does **not** work.
It has a readable attribute.

It could be argued, that it should not be readable. I would only expect, that it can be executed.

I wondered if selinux forbids reading, but selinux cannot be shown by ls:
```
# ls -AlZ /system/bin/su                                                                                                                                
-rwxr-xr-x 1 root shell ?  294688 2009-01-01 01:00 /system/bin/su
```
also `# ls -AlZ /system/bin/` does not show su (which can also be argued)
This prevents a search for it via listing the directory.
However, `which su` works, because it directly uses the file path (combining each PATH components with the explicit filename).
I think, these things need not be restricted, because only apps with root rights can see the file anyways.

btw. I wonder about #734, because 10763 did not allow me to use the su command in certain situations, e.g. in adb to elevate from the user to root.

closed 2025-03-09 by Admirepowered