bmax121/APatch #443: apatch faild to install modules，the return is : device or resource busy error 16

tobuusagi, 2024-05-11:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

修补boot开机后可以正常安装系统模块，但是装完一批模块重启后就再无法安装了，报错error16。

### Reproduce method/复现方法

安装任意模块

### Expected behavior/预期行为

正常安装模块

### Actual behaviour /实际行为

报错，提示error16

### Screenshots/截图

[image]

### Logs/日志

[APatch_bugreport_2024-05-11_22_27.tar.gz](https://github.com/bmax121/APatch/files/15283156/APatch_bugreport_2024-05-11_22_27.tar.gz)
[APatch_install_log_2024-05-11-22-23-17.log](https://github.com/bmax121/APatch/files/15283161/APatch_install_log_2024-05-11-22-23-17.log)

### Device Name/设备名称

Google Pixel 6

### OS Version/系统版本

安卓14稳定版以及安卓15beta1均有此问题

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

5.10.198-android13-4-00036-g11d7e4f57f68-ab11392938

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

[APatch_bugreport_2024-05-11_22_27.tar.gz](https://github.com/bmax121/APatch/files/15283172/APatch_bugreport_2024-05-11_22_27.tar.gz)
[APatch_install_log_2024-05-11-22-23-17.log](https://github.com/bmax121/APatch/files/15283173/APatch_install_log_2024-05-11-22-23-17.log)

pomelohan, 2024-05-11:
装了防格机？

tobuusagi, 2024-05-12:
没有，就这些模块，没了。
[image]
[image]

pomelohan, 2024-05-15:
日志不太够，可以尝试按照下面的步骤重新尝试一次吗：
1、触发安装失败
2、不要重启，立刻去 设置 -> 发送日志

tobuusagi, 2024-05-15:
[APatch_bugreport_2024-05-15_17_01.tar.gz](https://github.com/bmax121/APatch/files/15319142/APatch_bugreport_2024-05-15_17_01.tar.gz)
这是按你要求重新生成的日志

Pzufhegfe, 2024-06-04:
我pixel8也有这个问题
从安卓14到现在beta（ap31.240426.023.b4）都有
一般是开机后过一会就会出现这个问题
[image]
[image]

Pzufhegfe, 2024-06-04:
[APatch_bugreport_2024-06-04_15_36.tar.gz](https://github.com/user-attachments/files/15544243/APatch_bugreport_2024-06-04_15_36.tar.gz)
这是日志

tobuusagi, 2024-06-04:
我现在换了kernelsu开机一会也会有类似的问题，但是错误代码是1。不过只需要在安装界面选择直接安装，重启之后有一小段时间可以正常安装卸载模块。

获取Outlook for Android<https://aka.ms/AAb9ysg>

Screenshot_20240604-153303.png (view on web)<https://github.com/bmax121/APatch/assets/87434326/006d18ae-240e-4686-987b-1d2a2147c152>
Screenshot_20240604-153310.png (view on web)<https://github.com/bmax121/APatch/assets/87434326/12471526-ddae-48b5-8a5a-bdf19b71fcc8>

Pzufhegfe, 2024-06-04:
而且我用scene创建zram或者swap也会出现这个device or resource busy，不过再试一次一般就好了

pomelohan, 2024-06-05:
> 我pixel8也有这个问题 从安卓14到现在beta（ap31.240426.023.b4）都有 一般是开机后过一会就会出现这个问题 ![Screenshot_20240604-153303](https://private-user-images.githubusercontent.com/87434326/336350117-006d18ae-240e-4686-987b-1d2a2147c152.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc1NjIxNTYsIm5iZiI6MTcxNzU2MTg1NiwicGF0aCI6Ii84NzQzNDMyNi8zMzYzNTAxMTctMDA2ZDE4YWUtMjQwZS00Njg2LTk4N2ItMWQyYTIxNDdjMTUyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA1VDA0MzA1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJlYTM1YTIwZmZhNjllYWEwOGRmYzVhNmQ3NjI5YTZlODFhZDAzZTIwZWVjMzEwYTM5ODA2OGJiMjA2YWNiZDImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.PfDs3WRjxxzzD478HhiclNdwWFRtBhh7H9gyKU6mT_A) ![Screenshot_20240604-153310](https://private-user-images.githubusercontent.com/87434326/336350141-12471526-ddae-48b5-8a5a-bdf19b71fcc8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc1NjIxNTYsIm5iZiI6MTcxNzU2MTg1NiwicGF0aCI6Ii84NzQzNDMyNi8zMzYzNTAxNDEtMTI0NzE1MjYtZGRhZS00OGI1LThhNWEtYmRmMTliNzFmY2M4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA1VDA0MzA1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY1NWE1ZGEyZjAxNzE0Y2E0YWYwNWE2NDNlMzA1NWQxNTgwYzNkMzRmOWZlODFjOWM3NWEzYjFiYjViNDMyM2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.1ysbpTn67V84X4X7HRwwvQSqAuK_DH8a6SETSpQSx6E)

可以远程调试吗，ToDesk

> 而且我用scene创建zram或者swap也会出现这个device or resource busy，不过再试一次一般就好了

安装模块的时候再试一次可以好吗？

Pzufhegfe, 2024-06-05:
> 可以远程调试吗，ToDesk
>
>
> 安装模块的时候再试一次可以好吗？

不行，ksu也是，装过的模块也不能禁用，不能卸载，新的装不上

Pzufhegfe, 2024-06-05:
> 可以远程调试吗，ToDesk
>
>
> 安装模块的时候再试一次可以好吗？

远程的话可以，大概什么时间呢

Pzufhegfe, 2024-06-07:
突然发现好像是zygisk的原因，我重刷apatch后没装zygisk，到目前为止都没有重现这个问题。而且我的设备装了zygisk会导致频繁的自动重启（大概两三个钟头一次）

pomelohan, 2024-06-17:
最新CI版还能复现？

Pzufhegfe, 2024-06-23:
> 最新CI版还能复现？
前几天把手机恢复出厂然后就不会了，现在在用ksu反正不会，感觉是ota导致的

closed 2024-06-23 by pomelohan