bmax121/APatch #447: ARCore related modules will cause mobile data to be disconnected from the network | ARCore 模块无法正常工作

Sostay, 2024-05-15:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

之前一直通过Magiks使用这个模块 https://github.com/Magisk-Modules-Repo/ARCore_enabler，在未通过Google认证的机型上使用AR应用，但是换到APatch之后，刷入并启用模块后移动数据无法联网。

### Reproduce method/复现方法

1.通过APatch 刷入 https://www.celsoazevedo.com/files/random/ARCore_Playground_Patcher-v2.1.zip
2.重启手机，关闭WiFi，打开移动数据，无法连接网络。

### Expected behavior/预期行为

ARCore验证通过，可以正常使用AR相关的应用

### Actual behaviour /实际行为

移动数据连不上

### Logs/日志

[APatch_bugreport_2024-05-15_21_37.tar.gz](https://github.com/bmax121/APatch/files/15322330/APatch_bugreport_2024-05-15_21_37.tar.gz)

### Device Name/设备名称

xiaomi MIX4

### OS Version/系统版本

Hyper OS 1.0.2

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

5.4.233

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

虽然我知道这个模块已经很久没有更新维护了，但是一直找不到替代方案，之前在Magisk上也能正常工作，所以想在此寻求一个在APatch能正常工作的兼容性方案。

希望有线索的人能提供一些帮助，感谢！

pomelohan, 2024-05-18:
不刷入模块可以使用移动网络上网吗

Sostay, 2024-05-18:
> 不刷入模块可以使用移动网络上网吗

可以的 反复测试确认过了，只要刷入这个模块 就上不了网，禁用重启之后一切正常。而且这个模块开启之后它之前的AR相关的功能也完全可以正常工作~

pomelohan, 2024-05-20:
所以这就不是 APatch 应该修复的了，如果你不刷入模块出现网络问题，才是应该修复的，这个你需要找模块的作者进行适配。

closed 2024-05-20 by pomelohan