bmax121/APatch #451: System modules not validated!

MIkeeJY, 2024-05-17:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

[image]

### Reproduce method/复现方法

模块安装完不生效，始终显示灰色，卸载也不生效

### Expected behavior/预期行为

模块可以正常开启关闭

### Actual behaviour /实际行为

模块灰色 并且不生效

### Screenshots/截图

[image]

### Logs/日志

[APatch_bugreport_2024-05-17_19_26.tar.gz](https://github.com/bmax121/APatch/files/15348463/APatch_bugreport_2024-05-17_19_26.tar.gz)

### Device Name/设备名称

CAIC

### OS Version/系统版本

13

### APatch Version/APatch 版本

APatch_10763_10763-release-signed

### Kernel Version/内核版本

5.15

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

[APatch_bugreport_2024-05-17_19_33.tar.gz](https://github.com/bmax121/APatch/files/15348495/APatch_bugreport_2024-05-17_19_33.tar.gz)

pomelohan, 2024-05-17:
你这明显不是 10763 版本

MIkeeJY, 2024-05-18:
@pomelohan 大佬，结尾的log是升级到最新版本的log哦 依然不可以 跟截图一样，我特意附了两个log，一个是旧版本的，一个是更新到最新以后得log，望看下哦

pomelohan, 2024-05-18:
duplicated #344. Closed as not planned

MIkeeJY, 2024-05-26:
@bmax121 hi,the log for this bug,the rom is customed,please help me to solve it,thx a lot <3
[APatch_bugreport_2024-05-20_10_13.tar.gz](https://github.com/bmax121/APatch/files/15446464/APatch_bugreport_2024-05-20_10_13.tar.gz)

MIkeeJY, 2024-05-27:
[
  {
    "version": "1.0.5 (305-d463cbb-release)",
    "id": "zygisksu",
    "enabled": "true",
    "remove": "false",
    "author": "Nullptr, 5ec1cff",
    "name": "Zygisk Next",
    "web": "false",
    "update": "true",
    "description": "Standalone implementation of Zygisk.",
    "updateJson": "https://api.nullptr.icu/android/zygisk-next/static/update.json",
    "versionCode": "305"
  },
  {
    "web": "false",
    "versionCode": "7244",
    "remove": "false",
    "version": "v1.9.3_mod (7244)",
    "enabled": "true",
    "id": "zygisk_lsposed",
    "author": "LSPosed Developers",
    "name": "Zygisk - LSPosed",
    "update": "true",
    "description": "Another enhanced implementation of Xposed Framework. Supports Android 8.1 ~ 14. Requires Magisk 24.0+ and Zygisk enabled.",
    "updateJson": "https://mywalkb.github.io/LSPosed_mod/release/zygisk.json"
  }
]    对比了下真机 update这个真机是false， 但是在云手机这个值是true

pomelohan, 2024-06-14:
你这个是云手机？我能调试吗

pomelohan, 2024-06-14:
> @pomelohan 你Q有验证我加不了，可以留个联系方式 todesk远程看下嘛？

t.me/real1f2003d5

closed 2024-06-15 by MIkeeJY