bmax121/APatch #455: kernel config and apatch compatibility issue

xxjoe2, 2024-05-18:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [ ] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

with apatch version 0.10.7 (10763) on my phone (linux 4.9), ap was running without problem. then i installed another kernel with different config with required configs set: 
`
CONFIG_KALLSYMS=y
CONFIG_KALLSYMS_ALL=y
CONFIG_KERNEL_DEBUG=y (this is needed or otherwise CONFIG_KALLSYMS_ALL will be discarded)
`

ap patching process was a success, but after installed the pathch kernel, i found some of the apps, which have been granted with root access, did not have root access anymore. i have checked package config file and the uids were correct.

then i opened terminal emulator (granted with root access) and issused su but got an error

### Reproduce method/复现方法

$ su
/system/bin/sh: /system/bin/su: No such file or directory

### Expected behavior/预期行为

expected root access

### Actual behaviour /实际行为

$ su
/system/bin/sh: /system/bin/su: No such file or directory

however, i found that it was due to the shell (sh). after i switched to another shell root was granted
`
busybox sh
su
`
so i made a module and put a symlink /system/bin/sh and pointed it to busybox (ie sh -->busybox). after the phone rebooted, all root accesses were granted properly

so my question are:
1. could it be a bug?
2. except CONFIG_KALLSYMS and CONFIG_KALLSYMS_ALL, i bet there are some more configs which affect the ap, and obviously CONFIG_KERNEL_DEBUG is one of them in my case. can you name any other configs which may cause ap not to work properly? for instance other debug config, etc

### Device Name/设备名称

samsung note 9

### OS Version/系统版本

android 13

### APatch Version/APatch 版本

0.10.7 (10763)

### Kernel Version/内核版本

4.9

### KernelPatch Version/KernelPatch 版本

n/a

### Additional context/其他信息

n/a

Ai686Leo, 2024-05-23:
> oh btw, there are 2 custom kernels for my device. one of them works properly after patching. but this kernel, although i have both CONFIG_KALLSYMS and CONFIG_KALLSYMS_ALL set to yes, it just didnt work as expected. so may i ask, did you know if there are any other kernel configs which may affect apatch?

Your method is great too (sorry just saw that you've solved it yourself). Whether other configurations affect Apatch or not, I don't know 😓

xxjoe2, 2024-05-23:
actually it's not solved completely, the trick only helps in command line but it doesnt work for the apps with one of the kernels

as you mentioned, it is similar to issue #439, may be PATH issue in ap? idk, just wild guess

anyway, hope that it can be fixed soon

thanks a lot

closed 2025-03-09 by Admirepowered