bmax121/APatch #468: Root checked！

MIkeeJY, 2024-05-24:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Apatch 相关root被检测

### Reproduce method/复现方法

我写了一个demo 关于单独检测apatch的root    https://github.com/MIkeeJY/Root-Checker4Apatch

检测框架来自 https://github.com/scottyab/rootbeer

[image]

### Expected behavior/预期行为

Root不应该被检测到

### Actual behaviour /实际行为

Apatch相关Root被检测

### Screenshots/截图

[image]

### Device Name/设备名称

pixel6

### OS Version/系统版本

13

### APatch Version/APatch 版本

APatch_10763_10763-release-signed

### Kernel Version/内核版本

5.10

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

请看描述

marat2509, 2024-05-24:
HyperOS 1.0.3 Xiaomi.eu:

<details><summary>No root detected</summary>

[image]

[image]

[image]
</details>

MIkeeJY, 2024-05-25:
@marat2509 
[image]
[image]
[image]

pomelohan, 2024-05-25:
> @marat2509 [...]

Did you try removing all modules and whether could be detected?

MIkeeJY, 2024-05-25:
@marat2509 Where can I get lSPosed 1.9.3?

marat2509, 2024-05-26:
> @marat2509 Where can I get lSPosed 1.9.3? From https://github.com/pumPCin/LSPosed ?

Why not?

bmax121, 2024-05-26:
@MIkeeJY upload your bug report logs

MIkeeJY, 2024-05-26:
@bmax121 At last，I know the reason that why it can be checked，
[image]
[image]
  public boolean checkForRWPaths() {

        boolean result = false;
       
        //Run the command "mount" to retrieve all mounted directories
        String[] lines = mountReader();

        if (lines == null){
            // Could not read, assume false;
            return false;
        }

        //The SDK version of the software currently running on this hardware device.
        int sdkVersion = android.os.Build.VERSION.SDK_INT;
        
           /**
             *
             *  In devices that are running Android 6 and less, the mount command line has an output as follow:
             *
             *   <fs_spec_path> <fs_file> <fs_spec> <fs_mntopts>
             *
             *   where :
             *   - fs_spec_path: describes the path of the device or remote filesystem to be mounted.
             *   - fs_file: describes the mount point for the filesystem.
             *   - fs_spec describes the block device or remote filesystem to be mounted.
             *   - fs_mntopts: describes the mount options associated with the filesystem. (E.g. "rw,nosuid,nodev" )
             *
             */

            /** In devices running Android which is greater than Marshmallow, the mount command output is as follow:
             *
             *      <fs_spec> <ON> <fs_file> <TYPE> <fs_vfs_type> <(fs_mntopts)>
             *
             * where :
             *   - fs_spec describes the block device or remote filesystem to be mounted.
             *   - fs_file: describes the mount point for the filesystem.
             *   - fs_vfs_type: describes the type of the filesystem.
             *   - fs_mntopts: describes the mount options associated with the filesystem. (E.g. "(rw,seclabel,nosuid,nodev,relatime)" )
             */
        
        for (String line : lines) {

            // Split lines into parts
            String[] args = line.split(" ");

            if ((sdkVersion <= android.os.Build.VERSION_CODES.M && args.length < 4)
                    || (sdkVersion > android.os.Build.VERSION_CODES.M && args.length < 6)) {
                // If we don't have enough options per line, skip this and log an error
                QLog.e("Error formatting mount line: "+line);
                continue;
            }

            String mountPoint;
            String mountOptions;

            /**
             * To check if the device is running Android version higher than Marshmallow or not
             */
            if (sdkVersion > android.os.Build.VERSION_CODES.M) {
                mountPoint = args[2];
                mountOptions = args[5];
            } else {
                mountPoint = args[1];
                mountOptions = args[3];
            }

            for(String pathToCheck: Const.pathsThatShouldNotBeWritable) {
                if (mountPoint.equalsIgnoreCase(pathToCheck)) {

                       /**
                         * If the device is running an Android version above Marshmallow,
                         * need to remove parentheses from options parameter;
                         */
                        if (android.os.Build.VERSION.SDK_INT > android.os.Build.VERSION_CODES.M) {
                            mountOptions = mountOptions.replace("(", "");
                            mountOptions = mountOptions.replace(")", "");

                        }
                    
                    // Split options out and compare against "rw" to avoid false positives
                    for (String option : mountOptions.split(",")){

                        if (option.equalsIgnoreCase("rw")){
                            QLog.v(pathToCheck+" path is mounted with rw permissions! "+line);
                            result = true;
                            break;
                        }
                    }
                }
            }
        }

        return result;
    }     

This method is useful for detecting if critical directories on an Android device are mounted with rw permissions, which can indicate root access. The detection is handled by parsing the mount command output, which varies depending on the Android version, and comparing the mount options against a list of paths that should not be writable.

closed 2024-05-26 by MIkeeJY