bmax121/APatch #469: /proc/sys/kernel/ftrace_enabled reading broken after installing APatch

nitanmarcel, 2024-05-24:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

When using BTF and tracing while having the kernel patches with APatch, reading from `/proc/sys/kernel/ftrace_enabled` throws an error regarding that the device was not found.

### Reproduce method/复现方法

1. Compile a kernel with BTF and tracing enabled
2. Patch the kernel with APatch
3. do `cat /proc/sys/kernel/ftrace_enabled`

### Expected behavior/预期行为

Without APatch (adb root), reading the nod works fine, it returns either 0 or 1 depending if tracing is enabled or not.

### Actual behaviour /实际行为

After patching the kernel with APatch this error raises when trying to read from it:

`no such device`

### Device Name/设备名称

Oneplus 6t

### OS Version/系统版本

Android 14

### APatch Version/APatch 版本

10798 from actions

### Kernel Version/内核版本

4.9.337

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

It's a small thing, but maybe this can raise other issues behind it. I suppose it doesn't hurt to take a look at what happens more exactly

pomelohan, 2024-06-16:
Sorry but there was no `/proc/sys/kernel/ftrace_enabled` in my device

nitanmarcel, 2024-06-16:
> Sorry but there was no `/proc/sys/kernel/ftrace_enabled` in my device

Yeah that's a bpf thing, more exactly is tied with kpropes. Looks like the way APatch uses KProbes breaks some things under the hood

https://www.kernel.org/doc/Documentation/trace/ftrace.txt

closed 2025-03-09 by Admirepowered