bmax121/APatch #471: Add option to patch raw image like magsik

Joly0, 2024-05-26:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

Magisk has the option, to install itself, to patch the payload.bin and the raw image that you can simply do a manual install of in most roms.
It would be great to have such an option in apatch aswell. I do not have any custom recovery and therefore cant flash a modified boot.img, also i would need to extract the boot.img first, which is a little hassle aswell. Patching the raw image file directly would make it probably a lot easier for people t ibstall apatch.

### Describe the solution you'd like/描述你想要的解决方案

Option in apatch to patch raw image file and maybe payload.bin (raw image highly preferred)

### Describe alternatives you've considered/描述您考虑过的备选方案

None

bmax121, 2024-05-30:
maybe you need a kptools ? (KernelPatch release)

Joly0, 2024-05-30:
> maybe you need a kptools ? (KernelPatch release)

I am not sure what you mean?

bmax121, 2024-06-03:
https://github.com/bmax121/KernelPatch/releases/download/0.10.7/kptools-linux

Joly0, 2024-06-04:
Still not sure how this is supposed to help me or answer my feature request?

bmax121, 2024-06-06:
sorry, I thought it was the kernel image. as for payload.bin, we don't have time to support it

closed 2024-06-06 by bmax121