bmax121/APatch #475: profile not supported - same apps in different profiles treated as one single app

xxjoe2, 2024-05-27:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

profile is not supported, apatch looks like relies on the package name only as key but appid is not used

### Reproduce method/复现方法

install an app in normal profile, then clone it in work profile and set permissions with both apps
they will be saved as one single entry in package_config with one of the appid only

### Expected behavior/预期行为

expected to have 2 entries saved with both package name and appid as unique key

for example
com.whatsapp uid 10130
com.whatsapp uid 90130

### Actual behaviour /实际行为

only one is saved

although both should have same permissions normally, it is not unusual to have a file explorer with root access in normal profile but without root access in work profile

### Device Name/设备名称

samsung note9

### OS Version/系统版本

a13

### APatch Version/APatch 版本

10740 (i have problems with newer versions so i rolled back to a previous one)

### Kernel Version/内核版本

4.9

### KernelPatch Version/KernelPatch 版本

0.10.6

### Additional context/其他信息

also i found that changes in ap app does not reflected in package_config sometimes, ie, changes not saved to the config file but in memory only

pomelohan, 2024-05-27:
You are right, it’s a logical bug that manager only treat package name as its symbol. I will solve it when I am free.
`also i found that changes in ap app does not reflected in package_config sometimes, ie, changes not saved to the config file but in memory only`
Also remember do not manually edit the package_config except using manager, we didn’t listen this file change from kernel side. If you manually edited it, you should reboot your device to make it worked.

xxjoe2, 2024-05-28:
hi

thanks for your prompt reply and quick action

i have just tested it but i found that i couldnt check the results by reading the package_config file, it was not updated at all. so i cloned the root explorer to another profile for checking root access

and i can say, for the root access, it works without any problem. i can set an app in different profile with different root access. good work

however there are some minor issues. i can understand that, dont worry. and actually it doesnt worry me neither coz i dont need such function atm but there might be some users who want this

okay the problems are:
1 the uids were incorrectly shown between to two profile apps once in the ap app. for example:
`
root explorer uid 100300 (main profile)
root explorer uid 950321 (work profile)

were shown as
root explorer uid 950321 (main profile)
root explorer uid 100300 (work profile)
`

2 the exclude switch didnt work properly. i switched one of them to on, then back to the main page or exited the app, the exclude switch turned back to off

3 i also tried to switch on exclude for both of them, but when i revisted the page they were off

so the exclude function for different profile might need to be reviewed again, but the root access is good so far

pomelohan, 2024-05-28:
> i found that i couldnt check the results by reading the package_config file, it was not updated at all

`i found that i couldnt check the results by reading the package_config file, it was not updated at all`

Sorry I couldn’t understand this, can you describe it again? Thank you!

xxjoe2, 2024-05-29:
with my 10740, when i change anything in the ap mgr, eg switch on exclude for whatsapp , i can see corresponding changes in package_config (changes will be saved in the file) like
whatsapp 100289  0 1
and thats why i was able to find profile app is not supported

but after installing your build, changes were not saved in the config file. after i changed root access or exclude option for root explorer in both normal and work profile, i went to check the config file to see if there were 2 entries for root explorer. however  no changes were saved:
eg
grep rootexplorer package_config

nothing returned. i opened the file and confirmed there was no entry for root explorer

xxjoe2, 2024-05-29:
just updated to 10763, 1064fb8
and here's an example of what i meant by incorrect uid problems with different profile apps

[image]

pomelohan, 2024-05-29:
> just updated to 10763, [1064fb8](https://github.com/bmax121/APatch/commit/1064fb8a73a1c97a809cc475b7d7d92ab1a3456b) and here's an example of what i meant by incorrect uid problems with different profile apps. in this case, same uids for 2 different profiles
> 
> ![Screenshot_20240530_040252](https://private-user-images.githubusercontent.com/83495134/334951259-a1d9df93-6ebc-4206-b447-7f3ebe05c24d.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcwMjAwMzIsIm5iZiI6MTcxNzAxOTczMiwicGF0aCI6Ii84MzQ5NTEzNC8zMzQ5NTEyNTktYTFkOWRmOTMtNmViYy00MjA2LWI0NDctN2YzZWJlMDVjMjRkLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTI5VDIxNTUzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTkxOWMyYTQ0NjVmM2VlNGQwZjAxOTI3NWM1ZDM0OTE5ZDE0MzE5NjhmNDJjMWM2YTk0NTYzMzY3ODc1OWI4MzUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ORZMvQGHkF51L_BVVUcpc8J37kK_Dhs9mlM-DHZvpBY)

I will fix when I am free as soon as possible.

pomelohan, 2024-05-30:
> just updated to 10763, [1064fb8](https://github.com/bmax121/APatch/commit/1064fb8a73a1c97a809cc475b7d7d92ab1a3456b) and here's an example of what i meant by incorrect uid problems with different [...]

https://github.com/bmax121/APatch/actions/runs/9296784882/artifacts/1550942931

try this build

xxjoe2, 2024-05-31:
does this one write the "rules" to package_config correctly? the last one you gave me didnt write anything and after a few attempts and reboots, it showed only the header  but nothing else

pomelohan, 2024-05-31:
> does this one write the "rules" to package_config correctly? the last one you gave me didnt write anything and after a few attempts and reboots, it showed only the header but nothing else

Yes I fixed， try it

xxjoe2, 2024-05-31:
installed new apk, removed config, rebooted
unfortunately, didnt work properly, missing incorrect rules

[image]
[image]

[image]
[image]

pomelohan, 2024-06-01:
> installed new apk, removed config, rebooted unfortunately, didnt work properly, missing incorrect rules
> 
> ![Screenshot_20240601_034638](https://private-user-images.githubusercontent.com/83495134/335714088-92485b5e-bdfa-4375-a570-ecce0959ac81.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcyNTMzNjMsIm5iZiI6MTcxNzI1MzA2MywicGF0aCI6Ii84MzQ5NTEzNC8zMzU3MTQwODgtOTI0ODViNWUtYmRmYS00Mzc1LWE1NzAtZWNjZTA5NTlhYzgxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjAxVDE0NDQyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdkZjdmMmJmMDlkYWViMTY2YmFiMjU3Mjk4M2NmODFiOGNkMmU2MjAwNjVmZDc1Nzk0MmRlMGU5NjdjZWIxYTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.CnnYgkuzDJTb_thaw2AeeTM9JwMd9pT0dLO-JHPILV8) ![Screenshot_20240601_034618_Root Explorer](https://private-user-images.githubusercontent.com/83495134/335714110-88caa5e8-cdb3-4b80-b313-beb006de550c.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcyNTMzNjMsIm5iZiI6MTcxNzI1MzA2MywicGF0aCI6Ii84MzQ5NTEzNC8zMzU3MTQxMTAtODhjYWE1ZTgtY2RiMy00YjgwLWIzMTMtYmViMDA2ZGU1NTBjLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjAxVDE0NDQyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTgxZjE5NjQyYzNhMTdiYTgxYTQ0NjNiNGM5YTE2NWRiNTAyYjAyZjE5ZDU0ZmEwNGE1MmQzMTUyZjk3N2FhYTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uYlnuaIa8EPOMaWj2eGA4fqSZCXDIX_obW9PI-4fsUU)
> 
> missing shell in config whatsapp should be 0,1 (main) and 1,0 (work profile) in config, ie the first 2 lines should be: `9510337 1,0
> 
> 10337 0,1 `
> 
> ![Screenshot_20240601_034857](https://private-user-images.githubusercontent.com/83495134/335714182-2d7bdd3e-9faf-49e8-af8e-746b5d70e633.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcyNTMzNjMsIm5iZiI6MTcxNzI1MzA2MywicGF0aCI6Ii84MzQ5NTEzNC8zMzU3MTQxODItMmQ3YmRkM2UtOWZhZi00OWU4LWFmOGUtNzQ2YjVkNzBlNjMzLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjAxVDE0NDQyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQxZjU2ZDhhMDI3ZDAyYTQ3OTMyOTFmYTM2OThhMTgwY2ZhNmY5MzMzM2NlODU0NTI2ZmY2MzAwODA1MDNjYzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.WV2RYdbXdMVAivaGjBh356FQQQ2JOBnQIh9k_yXa74E) ![Screenshot_20240601_034848_Root Explorer](https://private-user-images.githubusercontent.com/83495134/335714170-e786feb0-2400-42a7-b741-b7358f895ab1.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcyNTMzNjMsIm5iZiI6MTcxNzI1MzA2MywicGF0aCI6Ii84MzQ5NTEzNC8zMzU3MTQxNzAtZTc4NmZlYjAtMjQwMC00MmE3LWI3NDEtYjczNThmODk1YWIxLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjAxVDE0NDQyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ0YjdiNTYxNTUyYmMwMzNlM2QzNTJlMzM2NmFkNGQyMTcwMzc1ZmU0ZDM2OWZiZjJjMzljZDhhMmY4NWFiM2ImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.BSHr3OwpTFrSXU70h_jD-D_U5P9c2msFEVP_6RTjKMA)
> 
> shell appeared in config but not in app didnt change whatsapp but it became both allowed with root

1. `com.android.shell` is always in allow list from kernel, it doesn't matter if not displayed in manager or `package_config`
2. I'm trying my best to fix this bug since I can't reproduce it very exactly (I don't use work profile at all), so please be patient.
3. Try this and tell me result: https://github.com/bmax121/APatch/actions/runs/9331066388/artifacts/1558930023 (same operation as I mentioned before)
4. Thank you. 🌹

Admirepowered, 2024-06-01:
can you provide /data/system/packages.list file

xxjoe2, 2024-06-01:
i am backing up my data partition rn so cant test it until the backup is finished

but i can tell u there is no difference, ie only one single entry for the apps installed in the list, and no uids for different user profiles

btw i have some suggestions regarding this issue. actually i had a very difficult time after installing the first test build. for some reasons, after i restored the ap manager, the lsposed had some problems. it simply didnt load or active and a reinstall didnt help neither. i had to uninstall and reinstall the ap mgr and the modules, as well as lsposed. 

i didnt know what happen but it seems there are some compatibility issues like other users, and that's why i had to install an older version since once the new version was installed, my phone would not boot 3 out of 4 times

so i would like to ask if it is possible , how hard would it be, to sperate the ap mgr's app selection together with the write to config part so that i can test it easily? i mean just a fake app with a different package name and without actual granting them root accesses? i guess this would be quite easy, by just skipping the grant access process will do

if that will take you quite a lot of time, you may try to append some fake profile apps to the app list you get, for example com.taobao 950001, just for testing the result since you dont have dual profile

anyway these are just my 2 cents. what do you think?

xxjoe2, 2024-06-03:
hi

just tested, uids and root access worked properly

however there was one bug left.
steps:
1 set both with root access, correctly done
2 set one of them to exclude only, ie 1,0 and 0,1
3 config correctly written
4 close ap and reopen
5 ui and config file entry mismatch, ie both shows 0,1 in config

[image]
[image]

pomelohan, 2024-06-03:
> hi
> 
> just tested, uids and root access worked properly
> 
> however there was one bug left. steps: 1 set both with root access, correctly done 2 set one of them to exclude only, ie 1,0 and 0,1 3 config correctly written 4 close ap and reopen 5 ui and config file entry mismatch, ie both shows 0,1 in config
> 
> ![Screenshot_20240603_165229](https://private-user-images.githubusercontent.com/83495134/336016074-d9ef89d0-f688-45f1-a73a-a83f143e14b3.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0MjI0OTQsIm5iZiI6MTcxNzQyMjE5NCwicGF0aCI6Ii84MzQ5NTEzNC8zMzYwMTYwNzQtZDllZjg5ZDAtZjY4OC00NWYxLWE3M2EtYTgzZjE0M2UxNGIzLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjAzVDEzNDMxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA0MDY2NGNhNTgzMjAwNGRjYzJiMGJmNjAzNDY3OTRkYmMxYTNiMDJkNTAyYWExZTE1MDM2ZWMxNTQ0YmE2MWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.SEdMqlpZRi7pXXCmHIiROiIwoPJz7cmg8pjdlVALO5s) ![Screenshot_20240603_165140_Terminal Emulator](https://private-user-images.githubusercontent.com/83495134/336016090-8f249236-cd18-40a2-abce-f640e51ebab6.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc0MjI0OTQsIm5iZiI6MTcxNzQyMjE5NCwicGF0aCI6Ii84MzQ5NTEzNC8zMzYwMTYwOTAtOGYyNDkyMzYtY2QxOC00MGEyLWFiY2UtZjY0MGU1MWViYWI2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjAzVDEzNDMxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEyNzdiODY1ZWI3ZDU1MmFkZjgyNjM5N2VhNDk3ODA5OTE2YTY0N2NhOTczODkwNzk2MDE1Mjk0ZmMwY2M0ZjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.qBPwKQIDWWAnbY5bkVkH9bLQbVtG2Ch-v_XV2T09Se4)

Can you confirm whether step 3 is executed successfully?

Did you do anything else after opening ap manager in step 4 and it changed without any manual operation?

xxjoe2, 2024-06-05:
yes, i read the config file after changes every time to see if they are saved correctly

no i never edit the config file manually

pomelohan, 2024-06-05:
> yes, i read the config file after changes every time to see if they are saved correctly
> 
> no i never edit the config file manually

Can you pm me in Telegram if possible?

t.me/real1f2003d5

pomelohan, 2024-06-15:
Maybe fixed in 
https://github.com/bmax121/APatch/actions/runs/9528468081/artifacts/1604807636

closed 2024-06-17 by pomelohan

xxjoe2, 2024-06-17:
couldnt get root access after installing this
- modules were executed
- no su
- in root app list, command line was not checked initially
- enabled root access to cmd line, root explorer etc, selected menu -> referesh -> all selected apps were disabled, no root access at all
- rebooted to recovery,  a new package_config found, entries were added
- rebooted again, root access were allowed, changes in main/profile apps were correctly shown in package_config and displayed in app list
- added new root access to other apps, not granted until reboot