bmax121/APatch #483: Simcard setting goin inactive after installing some module

bobbyamarthaa, 2024-05-31:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Every time I install the module.
I found my simcard settings to be inactive.
That makes me unable to do any settings on the Simcard settings.

I am currently found several modules,
Some modules that can trigger happen.
Pixelify
Lawnchair Module
Zygisk-Detach, YT & Music Revanced by J-HC.

If I pay attention, all modules that modify the system/product/ will trigger the same problem.

sorry if not related.
but i try in lastest magisk with same module, everything is fine. 

### Reproduce method/复现方法

Install any module that leads to /system/product

### Expected behavior/预期行为

The module should have nothing to do with SIMCard settings.

### Actual behaviour /实际行为

can't change any simcard settings

### Screenshots/截图

[image]
[image]

### Device Name/设备名称

Nokia 8.1 (X7)

### OS Version/系统版本

Android 11 (stock)

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

4.9.227

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

[APatch_bugreport_2024-05-31_23_18.zip](https://github.com/user-attachments/files/15516380/APatch_bugreport_2024-05-31_23_18.zip)

pomelohan, 2024-06-14:
Can you reproduce it with latest CI?

bobbyamarthaa, 2024-06-20:
here we go. 

[APatch_bugreport_2024-06-20_10_56.tar.gz](https://github.com/user-attachments/files/15908282/APatch_bugreport_2024-06-20_10_56.tar.gz)

bobbyamarthaa, 2024-06-20:
> Can you reproduce it with latest CI?

tested with new CI got same problem
clean install stock firmware > patch boot img with lastest CI > intstall some module. 

here we go. 

[APatch_bugreport_2024-06-20_10_56.tar.gz](https://github.com/user-attachments/files/15908282/APatch_bugreport_2024-06-20_10_56.tar.gz)

pomelohan, 2024-06-21:
Thank you. Will check it later

bobbyamarthaa, 2024-07-04:
any update about this?

pomelohan, 2024-07-22:
try latest CI, should work now.

bobbyamarthaa, 2024-07-23:
> try latest CI, should work now.

wich version?
just try nighly build, its broke my baseband.

closed 2025-03-09 by Admirepowered