bmax121/APatch #495: Mi Note 10 Lite (toco) reboot on lock screen

devnoname120, 2024-06-08:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

After flashing the patched `boot.bin` and rebooting the device unexpectedly reboots when it reaches the lockscreen.

### Reproduce method/复现方法

1. Download PixelOS
2. Extract boot.bin
3. Patch it
4. Reboot to recovery
5. Flash the patched boot.bin
6. Reboot
7. Reach the lockscreen.
8. Input your lockscreen unlock key, if you're fast enough you will the home page. Otherwise it will reboot while you are in the middle of typing your lockscreen unlock key.

### Expected behavior/预期行为

I can use the device normally.

### Actual behaviour /实际行为

Forced reboot.

### Screenshots/截图

TODO

### Logs/日志

The bug report from current CI is empty: 
[APatch_bugreport_2024-06-08_21_21.tar.gz.zip](https://github.com/user-attachments/files/15749417/APatch_bugreport_2024-06-08_21_21.tar.gz.zip)

Here is the bug report when I tried with the latest release though: 
[APatch_bugreport_2024-06-08_21_08.tar.gz](https://github.com/user-attachments/files/15749421/APatch_bugreport_2024-06-08_21_08.tar.gz)

### Device Name/设备名称

Mi Note 10 Lite

### OS Version/系统版本

Android 14

### APatch Version/APatch 版本

CI

### Kernel Version/内核版本

5 or something

### KernelPatch Version/KernelPatch 版本

TBD

### Additional context/其他信息

N/A

gyrorayo, 2024-06-22:
That rom you're using, is not a GSI, right?

devnoname120, 2024-06-22:
No it's not. Unfortunately the kernel is too old for that and it would be a considerable amount of effort to upgrade it without manufacturer support.

gyrorayo, 2024-06-22:
I'm no developer, but  my guess is that, this is an issue with the ROM not supporting APatch and not the other way around, I'd say that the **boot.img** you're trying to patch is **already modified**. Maybe my suggestion is a little dumb, but, have you tried using other rom? IMO, for Xiaomi devices, we should stick to MIUI or (stock or custom by xiaomi.eu)

https://xiaomi.eu/community/threads/miui-12-0-12-1-12-2-12-5-stable-release.56191/

I'm sorry, as I already stated, this might be dumb and you REALLY want to use PixelOS, but I had a Redmi9T, used that same 12.5 rom, and it's heaven, man. Plus, never had problems with APatch on that rom.

devnoname120, 2024-06-22:
@gyrorayo KernelSU works without issues though. MIUI is bloated spyware, my phone is way snappier since I switched to an AOSP ROM.

gyrorayo, 2024-06-22:
I know, dude, xiaomi.eu is a heavily modified ver of MIUI, removes most of spyware/bloatware. Really nice rom, extends battery life, and stuff. 

Sorry for not being helpful here. Just thought it'd be an issue with **boot.img** already patched/modified

devnoname120, 2024-07-13:
@pomelohan Is there something (logs, etc.) I can do to help the investigation?

closed 2025-03-09 by Admirepowered