bmax121/APatch #500: 在Android源代码编译时集成apatch

uwu-gl, 2024-06-10:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

想在Android源码编译编译出boot镜像之前或者之后集成apatch，然后再签名boot镜像，使得可以在刷入带有apatch支持的系统后锁定bootloader
（虽然要是想更新apatch版本就必须重新出包比较蛋疼）

### Describe the solution you'd like/描述你想要的解决方案

提供一种方法或思路，直接在boot镜像生成时加进参数

### Describe alternatives you've considered/描述您考虑过的备选方案

我认为可以参考现有magisk vendor的解决办法，例如android_vendor_magisk

MIkeeJY, 2024-06-10:
同求啊~

weisuofayubc, 2024-06-11:
啥？你能拿到手机厂商源代码？

MUYU212, 2024-06-11:
> 啥？你能拿到手机厂商源代码？

安卓不是开源的吗

MIkeeJY, 2024-06-11:
@weisuofayubc AOSP源码了解下？

weisuofayubc, 2024-06-12:
刷了lineageOS这种再回锁BL不是变砖吗

uwu-gl, 2024-06-12:
> 刷了lineageOS这种再回锁BL不是变砖吗

刷入avb custom key

uwu-gl, 2024-06-12:
刚才思考了一下，apatch所做的修改可能只有以下几个地方:
第一是sepolicy，这个可以直接在单独存储库定义，只需要在设备特有树include一下就行；
第二是内核，这个还要问问bmax121大佬这个地方是怎么实现比较好。（此issue的目的了）
另外，还有一个问题是超级密钥怎么处理，一般超级密钥都是自己定义自己用的，这个估计得在源码里面写死秘钥了（总不能上传一个云端隔一段功夫就curl获取一下吧）

bmax121, 2024-06-13:
有源码为啥不用ksu

uwu-gl, 2024-06-13:
> 编译源码为啥不用ksu

非gki😭

uwu-gl, 2024-06-13:
ksu新版已经不支持，于是就在寻求新的方式
@bmax121

closed 2024-06-16 by pomelohan