bmax121/APatch #508: APatch granting root permission AUTOMATICALLY when INSTALLING apps

gyrorayo, 2024-06-12:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Apps like **Termux** or **Zarchiver Pro** are automatically added to Superuser list when installed from Play Store.
Didn't have the issue when installing via file manager (ReVanced Manager, AdAway)

### Reproduce method/复现方法

- 1. Install Termux or ZArchiver from Play Store (so far these are the apps I noticed were added automatically)
- 2. Open APatch Manager, go to Superuser tab
- 3. Newly installed apps are now automatically added to the Superuser ALLOW list

### Expected behavior/预期行为

Not adding newly installed apps to Superuser allow list automatically, user should determine whether or not granting superuser permission to apps, even if they do support root functions.

### Actual behaviour /实际行为

When installing apps such as ZArchiver Pro and Termux via Play Store, are automatically added to superuser ALLOW list.

### Device Name/设备名称

Redmi Note 13 4G (Sapphire)

### OS Version/系统版本

HyperOS 1.0.3 by xiaomi.eu

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

android13-5.15.123

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

I assume that the only app that should be allowed automatically is Shell (com.android.shell)

crystle01, 2024-06-14:
Autostarts is another app that was granted root without permission, installed from fdroid.

pomelohan, 2024-06-16:
Can you reproduce this bug on CI Version?

MIkeeJY, 2024-06-16:
@pomelohan  related with https://github.com/bmax121/APatch/issues/386

gyrorayo, 2024-06-16:
> Can you reproduce this bug on CI Version?

sorry, I'm being ignorant, what's CI ver? how can I get it to test?

pomelohan, 2024-06-16:
> > Can you reproduce this bug on CI Version?
> 
> sorry, I'm being ignorant, what's CI ver? how can I get it to test?

https://github.com/bmax121/APatch/actions/runs/9528468081/artifacts/1604807636

pomelohan, 2024-06-16:
> @pomelohan related with #386

I don't think it's the same problem

gyrorayo, 2024-06-16:
> > > Can you reproduce this bug on CI Version?
> > 
> > 
> > sorry, I'm being ignorant, what's CI ver? how can I get it to test?
> 
> https://github.com/bmax121/APatch/actions/runs/9528468081/artifacts/1604807636

Thanks, quick question, do I need to patch boot.img again? or just update manager?

pomelohan, 2024-06-16:
> > > > Can you reproduce this bug on CI Version?
> > > 
> > > 
> > > sorry, I'm being ignorant, what's CI ver? how can I get it to test?
> > 
> > 
> > https://github.com/bmax121/APatch/actions/runs/9528468081/artifacts/1604807636
> 
> Thanks, quick question, do I need to patch boot.img again? or just update manager?

1. Update Manager
2. Repatch boot image with new manager
3. Update AndroidPatch
4. Try

gyrorayo, 2024-06-16:
> 1. Update Manager
> 2. Repatch boot image with new manager
> 3. Update AndroidPatch
> 4. Try

Done, reflashed newly patched boot.img, updated AndroidPatch. Reinstalled Zarchiver and Termux, APatch DIDN'T granted permissions automatically, so it's solved!

Thank you so much for the support. I'm loving APatch btw.

pomelohan, 2024-06-16:
Thanks for supporting us🌹

closed 2024-06-16 by pomelohan

SuperKenVery, 2024-07-18:
@gyrorayo I think you forgot to reopen the issue after commenting :D

gyrorayo, 2024-07-19:
@SuperKenVery I tried to do that, but there's no option for reopen issue :(

OpM3TA, 2024-12-21:
Me as well many apps it's happening to