bmax121/APatch #510: 最新测试版无法修补

lxx12445, 2024-06-12:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

最新测试10763-65-g178e3db_10828-release-signed无法直接刷入，通过修补boot刷入成功但是无法授权也无法刷入zygisk next和lsposed，显示版本报错22

### Reproduce method/复现方法

更新
修补刷入

### Expected behavior/预期行为

正常修补刷入

### Actual behaviour /实际行为

没法正常修补

### Screenshots/截图

[image]

### Device Name/设备名称

k40s （munch）

### OS Version/系统版本

android12 miui13

### APatch Version/APatch 版本

APatch_10763-65-g178e3db_10828-release-signed

### Kernel Version/内核版本

4.19

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

[Uploading APatch_bugreport_2024-06-12_23_43.tar.gz.zip…]()

miuirussia, 2024-06-12:
Confirm it with different message

[image]

lxx12445, 2024-06-13:
使用刷入boot.img安装成功，但是默认su路径变为/system/bin/kp,需要修改为/system/bin/su才能获取root，重启后失效，su路径变回kp
[image]

lxx12445, 2024-06-13:
[image]
模块无法刷入，shell权限已取消
[image]

lxx12445, 2024-06-13:
日志
[APatch_bugreport_2024-06-13_10_26.tar.gz.zip](https://github.com/user-attachments/files/15812936/APatch_bugreport_2024-06-13_10_26.tar.gz.zip)

pomelohan, 2024-06-14:
应该修了，可以再试试。Zygisk Next 不能用是已知问题

closed 2024-06-14 by pomelohan