bmax121/APatch #525: Button to install APModule is missing

gyrorayo, 2024-06-24:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Hi, noticed this a while ago, I wanted to update a module, but couldn't find the install button, weird thing, as I could install several modules before this update. You can find the button in KPModule tab, though. 

I'm using latest CI ver.

### Reproduce method/复现方法

1. Go to APModule tab
2. Try to install APModule
3. There's no Install button

### Expected behavior/预期行为

On APModule tab, there should be a button to install AP Modules on the bottom-right corner of the screen.

### Actual behaviour /实际行为

There's no install button on APModule tab

### Screenshots/截图

[image]
[image]

### Logs/日志

[APatch_bugreport_2024-06-24_12_13.tar.gz](https://github.com/user-attachments/files/15959834/APatch_bugreport_2024-06-24_12_13.tar.gz)

### Device Name/设备名称

Xiaomi Redmi 23129RA5FL (Redmi Note 13 4G)

### OS Version/系统版本

HyperOS 1.0.3 by xiaomi.eu (Android 14)

### APatch Version/APatch 版本

10844

### Kernel Version/内核版本

5.15.123-android13-8-00044

### KernelPatch Version/KernelPatch 版本

0.11.0

### Additional context/其他信息

At some point, could install modules without any trouble. Button was always there.

gyrorayo, 2024-06-24:
Sorry, fixed it by updating: **10844 > 10849**

pomelohan, 2024-06-26:
Yes, this is a known issue but we are busy, not fixed yet. Thanks for reporting.

gyrorayo, 2024-06-29:
fixed in 10857

closed 2024-06-29 by gyrorayo