bmax121/APatch #533: 经常需要开机好几次才能正常进入系统

naniccc, 2024-06-29:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

开机/重启引导界面经常黑屏，或者进入系统输入开机密码后瞬间黑屏，重复开机/重启多次才能进入正常系统

### Reproduce method/复现方法

开机/重启

### Expected behavior/预期行为

开机一次就能进入系统

### Actual behaviour /实际行为

开机/重启引导界面经常黑屏，或者进入系统输入开机密码后瞬间黑屏，重复开机/重启多次才能进入正常系统

### Logs/日志

[APatch_bugreport_2024-06-29_17_03.tar.gz](https://github.com/user-attachments/files/16039474/APatch_bugreport_2024-06-29_17_03.tar.gz)

### Device Name/设备名称

OnePlus PJA110

### OS Version/系统版本

ColorOS14

### APatch Version/APatch 版本

10854

### Kernel Version/内核版本

5.15.123-android13

### KernelPatch Version/KernelPatch 版本

0.11.0

### Additional context/其他信息

[APatch_bugreport_2024-06-29_17_03.tar.gz](https://github.com/user-attachments/files/16039481/APatch_bugreport_2024-06-29_17_03.tar.gz)

pomelohan, 2024-06-29:
试试关闭 Wi-Fi 重启是不是就好了

naniccc, 2024-06-29:
> 试试关闭 Wi-Fi 重启是不是就好了

好像是好了，试了3次都正常开机

devnoname120, 2024-07-16:
@pomelohan Why disable Wi-Fi?

pomelohan, 2024-07-20:
> @pomelohan Why disable Wi-Fi?

Because it was caused by qcacld bti panic

naniccc, 2024-08-18:
@pomelohan 又遇到开机输密码黑屏，强制重启好久才开机，不确定是升级模块的原因还是Apatch的原因，附上日志
[APatch_bugreport_2024-08-18_16_20.tar.gz](https://github.com/user-attachments/files/16647696/APatch_bugreport_2024-08-18_16_20.tar.gz)

naniccc, 2024-08-18:
@pomelohan 又遇到开机输密码黑屏，强制重启好久才开机，不确定是升级模块的原因还是Apatch的原因，附上日志
[APatch_bugreport_2024-08-18_16_20.tar.gz](https://github.com/user-attachments/files/16647705/APatch_bugreport_2024-08-18_16_20.tar.gz)

linnaikesi, 2024-08-29:
原来不只是我一个人会这样。重启之后进不去系统就卡住，然后又重启。得用一下救砖，进系统之后再一个一个重新加载模块

closed 2025-03-09 by Admirepowered