bmax121/APatch #543: 4.14.186 kernel irreparability boot

LX916-CN, 2024-07-11:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

4.14.186 kernel irreparability boot😭😭

### Reproduce method/复现方法

4.14.186 kernel irreparability boot

### Expected behavior/预期行为

4.14.186 kernel irreparability boot

### Actual behaviour /实际行为

4.14.186 kernel irreparability boot

### Screenshots/截图

[image]

### Logs/日志

>
[APatch_bugreport_2024-07-11_17_39.tar.gz](https://github.com/user-attachments/files/16174532/APatch_bugreport_2024-07-11_17_39.tar.gz)

### Device Name/设备名称

Redmi note 9 5G

### OS Version/系统版本

11

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

4.14.186-gdc06613e1adf

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

[Uploading APatch_bugreport_2024-07-11_17_39.tar.gz…]()

YUAN5711, 2024-07-12:
失败鉴权密码输入然后刷入修补

malikzaidimran, 2024-07-14:
i have same issue i have also this kernel.

YUAN5711, 2024-07-14:
How can we solve this problem?

aawja, 2024-07-15:
这个机型有没有安卓12？有的话升上去
[image]

YUAN5711, 2024-07-16:
> 这个机型有没有安卓12？有的话升上去
> Screenshot_2024-07-16-02-07-26-657_afh.gdf.ff.jpg (view on web)

谢谢

源

jia0826, 2024-09-07:
我的是红米note.11内核4.14.186刷完修完apatch修好的文件后重启设备不生效

closed 2025-03-09 by Admirepowered