bmax121/APatch #544: Request the ability to rename apk packages`s name

RASBI-CARDNAME, 2024-07-12:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

Some apps seem to detect the apatch manager to detect rooting, and if I uninstall the apatch manager app, it doesn't detect it.

### Describe the solution you'd like/描述你想要的解决方案

I wish there was a way to rename an app package like magisk to make it look like a different app

### Describe alternatives you've considered/描述您考虑过的备选方案

I uninstalled the app

crypticc1, 2024-09-01:
MS apps already detecting aPatch and so I also uninstall. Would be good for this not to be required without needing to resort to more modules and HMA

Ranieri92, 2024-09-05:
I need too just rename apk option

closed 2025-03-09 by Admirepowered