bmax121/APatch #551: Help!!! APatch 10763 with Kernel 4.4.146

DerBaliner, 2024-07-18:
Hi,

I need help!!!

I have successfully installed APatch 10763 on my UMIDIGI A5 Pro with kernel 4.4.146 and patched the boot image and successfully set the super key.

After I flashed the patched boot image with TWRP in the boot partition and restart the device, it reboots when the boot splash of the start screen appears.

I have successfully booted into the safe boot loop mode and get to the start screen, after a short time the device restarts.

Could it be that it has something to do with the kernel configuration? 

https://github.com/bmax121/APatch?tab=readme-ov-file#requirement

[APatch_bugreport_2024-07-17_17_41.tar.gz](https://github.com/user-attachments/files/16268338/APatch_bugreport_2024-07-17_17_41.tar.gz)

[image]
[image]
[image]
[image]
[image]

_Originally posted by @DerBaliner in https://github.com/bmax121/APatch/discussions/547_

DerBaliner, 2024-07-18:
I have now tried with version 0.10.7 and 0.11.0-dev, both boot without bootloop, but when I enter the super key, authentication fails again 🤔
https://kernelpatch-demo.pages.dev/

gyrorayo, 2024-07-19:
Device/kernel not compatible?
Have you tried with your stock rom?

DerBaliner, 2024-07-19:
> Device/kernel not compatible? Have you tried with your stock rom?

CONFIG_KALLSYMS_ALL=n
I have use from stock

A nice guy from XDA say to me, find kernel sources for mt6763
https://xdaforums.com/t/2024-full-guide-a14-qpr-3-apatch-ksu-microg-install-gsis-on-nabu-by-exam-with-leos-u-base-on-ponses-aosp-risingos-gsi.4675386/page-3

gyrorayo, 2024-07-19:
Yes, I've read  the thread, he gave you the steps to configure and compile your kernel, not a problem with APatch

DerBaliner, 2024-07-19:
> Yes, I've read the thread, he gave you specific steps to configure and compile your kernel, this issue is not a problem with APatch

Yep, I know
I close the issue and thx 😉

closed 2024-07-19 by DerBaliner