bmax121/APatch #553: /home/runner/work/KernelPatch/KernelPatch/tools/common.c:43/read_file_align(); open file kernel-b  - No such file or directory

sieger707, 2024-07-19:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

This is on Linux . I got a segmentation error 139 using automatic method of patching via 2 android phones ( P1 : same as boot.img P2 : copied boot.img to see if things change ) so I tried manual patching via linux . This is what happens . if there's a debug flag you want me to run I will do that and revert : 
[032  gh kernel
linwip/apatch_root_su # !1829
./kptools-linux -p --image kernel-b --skey  "-------somepassfdfdf#" --kpimg kpimg-android --out kernel
[-] /home/runner/work/KernelPatch/KernelPatch/tools/common.c:43/read_file_align(); open file kernel-b
 - No such file or directory
linwip/apatch_root_su # ls -altr
total 130236
drwxr-xr-x 1 root root    16384 Aug  2  2021 glibc-2.34
-rwxr-xr-x 1 root root  2044915 May 23 17:36 glibc-2.34-100.el9_4.2.x86_64.rpm
-rwxr-xr-x 1 root root      108 Jul 18 07:07 super.txt
-rwxr-xr-x 1 root root   105176 Jul 18 08:17 kptools-linux
-rwxr-xr-x 1 root root   171808 Jul 18 08:17 kpimg-android
-rwxr-xr-x 1 root root 13683566 Jul 18 08:55 magiskboot-linux-main.zip
drwxr-xr-x 1 root root        0 Jul 18 09:13 magiskboot-linux-main
-rwxr-xr-x 1 root root 41943040 Jul 18 09:16 wp15stock.img
-rwxr-xr-x 1 root root 36407296 Jul 18 09:19 kernel_b
-rwxr-xr-x 1 root root 21456384 Jul 18 09:19 ramdisk.cpio
-rwxr-xr-x 1 root root   150257 Jul 18 09:19 dtb
-rwxr-xr-x 1 root root 17301232 Jul 18 09:27 glibc-2.34.tar.xz
drwxr-xr-x 1 root root     4096 Jul 18 10:12 lib64
drwxr-xr-x 1 root root        0 Jul 18 10:12 etc
drwxr-xr-x 1 root root        0 Jul 18 10:12 var
drwxr-xr-x 1 root root        0 Jul 18 10:12 usr
drwxr-xr-x 1 root root        0 Jul 18 10:12 sbin
-rwxr-xr-x 1 root root        0 Jul 18 10:39 events.log
drwxr-xr-x 1 root root    61440 Jul 19 05:23 ..
drwxr-xr-x 1 root root     4096 Jul 19 07:02 .](url)
I already extracted `kernel binary` using `magiskboot` 

### Reproduce method/复现方法

as indicated above

### Expected behavior/预期行为

should have boot.img which is patched

### Actual behaviour /实际行为

above error

### Device Name/设备名称

Boot image from WP15 Oukitel. Compile on Linux

### OS Version/系统版本

Linux puppypc25224 5.17.1-lxpup64 #1 SMP PREEMPT Mon Mar 28 17:21:49 BST 2022 x86_64 Intel(R) Core(TM) i3 CPU       U 380  @ 1.33GHz GenuineIntel GNU/Linux

### APatch Version/APatch 版本

Latest

### Kernel Version/内核版本

4.14 for android boot.img and 5.17.1 for Linux Platform

### KernelPatch Version/KernelPatch 版本

Latest

### Additional context/其他信息

Just the error I get above

SakuraKyuo, 2024-08-04:
You need to rename kernel which unpacked to kernel-b

sieger707, 2024-08-06:
Segmentation fault. I got the same error in 2 Androids and now this one  on Linux. Are there any further debug flags I can add to tell you whats going on  
+ ./kptools-linux -p --image kernel_b --skey 'OkIScratchedItOff' --kpimg kpimg-android --out kernel
[+] kernel image_size: 0x022b8800
[+] kernel uefi header: false
[+] kernel load_offset: 0x00080000
[+] kernel kernel_size: 0x02b5b000
[+] kernel page_shift: 12
[+] new kernel image ...
[+] linux_banner 1: Linux version 4.14.186 (nobody@android-build) (Android (6443078 based on r383902) clang version 11.0.1 (https://android.googlesource.com/toolchain/llvm-project b397f81060ce6d701042b782172ed13bee898b79), LLD 11.0.1 (/buildbot/tmp/tmp6_m7QH b397f81060ce6d701042b782172ed13bee898b79)) #1 SMP PREEMPT Tue Nov 21 12:11:11 CST 2023
[+] linux_banner offset: 0x186b8ff
[+] kernel version major: 4, minor: 14, patch: 186
[+] kallsyms_token_table offset: 0x0160bc00
[+] endian: little
[+] kallsyms_token_index offset: 0x0160c000
[+] find arm64 relocation kernel_va: 0xffffff8008081000
[+] find arm64 relocation table range: [0x01b487d8, 0x02045260), count: 0x0003531b
[+] apply 0x0003531a relocation entries
[+] kallsyms_markers range: [0x0160b100, 0x0160bbb8), count: 0x00000157
[?] find approximate kallsyms_offsets error
[+] approximate kallsyms_addresses range: [0x013421d8, 0x013ec568) count: 0x00015472
[?] mismatch relo applied, subsequent operations may be undefined
[+] kallsyms_names offset: 0x013ed700
[-] /home/runner/work/KernelPatch/KernelPatch/tools/kallsym.c:442/find_num_syms(); kallsyms_num_syms error
[+] find arm64 relocation kernel_va: 0xffffff8008081000, but try use: ffffff8008080000
[+] find arm64 relocation table range: [0x01b487d8, 0x02045260), count: 0x0003531b
[+] apply 0x0003531a relocation entries
[+] kallsyms_markers range: [0x0160b100, 0x0160bbb8), count: 0x00000157
[?] find approximate kallsyms_offsets error
[+] approximate kallsyms_addresses range: [0x01342200, 0x013ed568) count: 0x0001566d
[?] mismatch relo applied, subsequent operations may be undefined
[+] kallsyms_names offset: 0x013ed700
[+] kallsyms_num_syms offset: 0x013ed600, value: 0x0001566d
[?] no linux_banner? maybe CONFIG_KALLSYMS_ALL=n?
[+] names table vector index: 0x00000012, pid_vnr index: 0x00001164
[+] pid_vnr verfied succeed, sp, insn: 0xd11083ff
[+] vectors offset: 0x00610000
[+] pid_vnr offset: 0x00791c48
[+] kallsyms_addresses offset: 0x01362af0
[+] layout kimg: 0x0,0x22b8800, kpimg: 0x22b9000,0x29f20, extra: 0x22e2f20,0x80, end: 0x22e2fa0, start: 0x2b5b000
[+] kpimg version: b00
[+] kpimg compile time: 13:55:04 Jul 14 2024
[+] kpimg config: android, release
[+] tcp_init_sock: type: T, offset: 0x00d119c0
[+] map_start: 0xd119c0, max_size: 0x800
[+] kallsyms_lookup_name: type: T, offset: 0x00272f90
[+] printk: type: T, offset: 0x0021ca00
[+] memblock_reserve: type: T, offset: 0x0034b038
[+] memblock_free: type: T, offset: 0x0034afc0
[+] memblock_mark_nomap: type: T, offset: 0x0034b4dc
[?] no symbol: memblock_phys_alloc_try_nid
[+] memblock_virt_alloc_try_nid: type: T, offset: 0x1bb479ac
[+] memblock_alloc_try_nid: type: T, offset: 0x5b2f499a
[+] panic: type: T, offset: 0x00147a34
[+] rest_init: type: t, offset: 0x00d15088
[?] no symbol: kernel_init
[+] kernel_init -> kernel_init$e6dd4e7ddeac37f8bcebf67a452b9410: type: t, offset: 0x00d1508c
[?] no symbol: report_cfi_failure
[?] no symbol: __cfi_slowpath_diag
[+] __cfi_slowpath: type: T, offset: 0x002cdc4c
[?] no symbol: copy_process
[+] copy_process -> copy_process.llvm.17179604929817515659: type: t, offset: 0x001462e0
[?] no symbol: avc_denied
[+] avc_denied -> avc_denied.llvm.2328894851612339589: type: t, offset: 0x005e8270
[+] slow_avc_audit: type: T, offset: 0x005e7e4c
[?] no symbol: input_handle_event
[+] superkey: OkIScratchedItOff
[+] paging_init: type: T, offset: 0x2e56746a
Segmentation fault

SakuraKyuo, 2024-08-15:
where's ur boot

sieger707, 2024-08-16:
https://mega.nz/file/ewU2TCBQ#dkFO1gCoOFgkkbqQZ21qB28L9RfEVtIHG-A_3eHPdKM
sorry  I should have given that before.

sieger707, 2024-08-17:
> https://mega.nz/file/ewU2TCBQ#dkFO1gCoOFgkkbqQZ21qB28L9RfEVtIHG-A_3eHPdKM
> sorry I should have given that before.
>
> https://mega.nz/file/J7UlmZQR#kSd9kdin4fb26Fvi--Rib4XPADZEQeExSknAD2z2G08
> Here's patched boot, but because of CONFIG_KALLSYMS_ALL is not set, so
> maybe it won't boot, so remember to backup ur origi boot image. SuperKey is
> abcd1234, it's for test

Thanks pal .. will it a shot 🧢

closed 2025-03-09 by Admirepowered