bmax121/APatch #556: APatch 10762 Pixel 3 XL - Android 11 bootloop and hang at lockscreen

bengalih, 2024-07-19:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Similar to:
https://github.com/bmax121/APatch/issues/423

but with APatch 10762 (latest release) bootloops. going back to 10657 boots fine.

### Reproduce method/复现方法

1. Installed Installed 10762 apatch and patch boot.img
2. Flash via fastboot flash boot boot.img
3. Phone reboots to lockscreen, is unresponsive, and reboots within a few seconds.
4. Process repeats until flashing back stock boot.img.
5. When trying same method with 10657, APatch seems to work fine.

### Expected behavior/预期行为

Normal boot of system with APatched image

### Actual behaviour /实际行为

Bootloop at lockscreen

### Logs/日志

I am unsure of which logs or how to gather, please direct me to a post which explains or provide info and I will get you what I can.

### Device Name/设备名称

Pixel 3 XL

### OS Version/系统版本

Android 11

### APatch Version/APatch 版本

10762

### Kernel Version/内核版本

4.9.248-gc4689af91bc5-ab7425221

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

I don't know what logs to send since I can't access anything.  Please let me know what is needed.

lxwang42, 2024-07-20:
i dont know why but this kernel version does not have CONFIG_KALLSYMS_ALL flag, that might be the reason.

bengalih, 2024-07-21:
I just wanted to add that I have also tried this with the latest 10763 and the result is the same.
Additionally, I normally use a custom Kernel instead of the stock which exhibits the behavior.
This is my custom kernel:

4.9.278-Kirisakura-R-Release-3.1.0+

And when applying 10763 Apatch I get the same results - reboot loop on lockscreen.

**_10657 however works on both Kernels mentioned._**

Output on the custom kernel:
`crosshatch:/ # uname -r
4.9.278-Kirisakura-R-Release-3.1.0+
crosshatch:/ # zcat /proc/config.gz | grep CONFIG_KALLSYMS
CONFIG_KALLSYMS=y
CONFIG_KALLSYMS_ALL=y
# CONFIG_KALLSYMS_ABSOLUTE_PERCPU is not set
CONFIG_KALLSYMS_BASE_RELATIVE=y
crosshatch:/ #`

After installing successfully from 10657, I update APatch to 10763 and am able to reboot without an issue.
However, after doing a direct in-place update from KernelPatch to 0.10.5 -> 0.10.7, I am again stuck in a bootloop.

Therefore, it would appear the issue is with version of 0.10.7 of KernelPatch.  

Does this mean I should open up a bug report with KernelPatch Instead?

20171213, 2024-07-24:
蛙趣，我昨天也遇到了小米14卡米logo，我以为是lsposed模块的原因，直接刷回了官方原版boot，但还是卡米无奈放弃数据直接线刷清除数据了，不知道是不是这个的原因，我用的是apatch 年7月22日的nightly版

grnassar, 2024-09-19:
I have a Pixel 3A XL (so somewhat different build), but I am using the stock kernel and am seeing the same bootloop on 10763.

Manual patching of the kernel with KernelPatch 0.11.1-dev instead of 0.10.7 successfully avoids the bootloop and boots properly; since I still have the 10763 apk (I have not had a chance to build the dev version manually and can't find it for dl) Apatch does successfully accept the superkey but su functionality predictably doesn't work.

Will try 10657 and see if that works as well, to match @bengalih 's findings.

closed 2025-03-09 by Admirepowered