bmax121/APatch #568: 自定义su命令

1q23lyc45, 2024-07-28:
### Is your feature request related to a problem?/你的请求是否与某个问题相关？

有些软件检测root时，使用su命令。如果允许换成别的，比如suu，8u或su1，能避免很多检测。

另外有些极特殊系统内核检测su命令，换成其他的命令（比如suu，timesu，或是sudo）才能执行。（比如，vivo手机Android 7.0-11.0的内核，理论上支持APatch，但是被这个su命令挡在门外了）

### Describe the solution you'd like/描述你想要的解决方案

修补页面上，在超级密钥输入框下面再加一个可选的输入框，能自定义su命令，比如说我输入suu修补出来的boot.img，其他app就必须用/system/bin/suu这个命令获取root权限，同时/system/bin/su自动失效。

(可能用到的：APatch App设置里面增加超级密钥鉴权执行命令，以适配自定义的su命令。)

### Describe alternatives you've considered/描述您考虑过的备选方案

https://github.com/4accccc/vivo-Magisk-Delta-suu

不过，bug也太多了。就连MagiskHide都是半残。

xiangxiangxiong9, 2024-08-02:
这个不行吗？[image]

tufeigunchu, 2024-08-02:
没有授权的app根本就看不到/system/bin/su

1q23lyc45, 2024-08-02:
> 这个不行吗？ [虽然但是，如果超级用户里没有勾选的话应该不会挂载su…] ![IMG_20240802_083359.jpg](https://private-user-images.githubusercontent.com/60531086/354421548-76368183-daf3-4dc0-97ed-138b9a6e2a17.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI1NjUzODIsIm5iZiI6MTcyMjU2NTA4MiwicGF0aCI6Ii82MDUzMTA4Ni8zNTQ0MjE1NDgtNzYzNjgxODMtZGFmMy00ZGMwLTk3ZWQtMTM4YjlhNmUyYTE3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODAyVDAyMTgwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWExYjFkODJkYzI0ZDA2NGFjNDEzYzAxNDFjZmFkZGQyMTU1NTNiNTc0ODNlZDM3NTQzMmIxZmNkYjhjYzk3ZjUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Sn74yJw-SR7AKl786OSaCc9PJdEXB0lt9GujhpnbfxY)

好像不行，vivo android 10只要是结尾带su就不让执行，bad interpreter: Operation not permitted。根本超级密钥都输入不了，输入完直接闪退了。那个选项输入超级密钥之前不显示。bug同 #323 。

xiangxiangxiong9, 2024-08-02:
> 好像不行，vivo android 10只要是结尾带su就不让执行，bad interpreter: Operation not permitted。根本超级密钥都输入不了，输入完直接闪退了。那个选项输入超级密钥之前不显示。bug同 #323 。

那……你有Rec吗？
有的话直接改 /data/adb/ap/su_path

1q23lyc45, 2024-08-02:
> 那……你有Rec吗？ 有的话直接改 /data/adb/ap/su_path

没有rec。我正在编译twrp呢，刚编译出来一个，触控不了，缺驱动文件，才补上还不一定能用。等编译完成看看

xiangxiangxiong9, 2024-08-02:
> 没有rec。我正在编译twrp呢，刚编译出来一个，触控不了，缺驱动文件，才补上还不一定能用。等编译完成看看

触控不行要是有电脑能连adb也行
echo 目标su路径 > /data/adb/ap/su_path

1q23lyc45, 2024-08-02:
先等等编译吧。那个半成品twrp，连adb都不管用

1q23lyc45, 2024-08-03:
可以了，谢谢

closed 2024-08-03 by 1q23lyc45

FrankHan052176, 2025-02-17:
我先用magisk suu实现root后，执行echo 目标su路径 > /data/adb/ap/su_path之后，再用apatch修补boot，开机依旧不行