bmax121/APatch #580: Unable to patch 4.14.141

a1489632, 2024-08-02:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

修补不了这个内核。
```
Linux localhost 4.14.141+ #1 SMP PREEMPT Thu Nov 30 17:24:16 CST 2023 aarch64 Android
```
对 `CONFIG_KALLSYMS_ALL=n` 的支持好像不完善
```
~ # zcat /proc/config.gz|grep CONFIG_KALLSYMS
CONFIG_KALLSYMS=y
# CONFIG_KALLSYMS_ALL is not set
# CONFIG_KALLSYMS_ABSOLUTE_PERCPU is not set
# CONFIG_KALLSYMS_BASE_RELATIVE is not set
~ #
```

另外，我无法使用最新CI版本！闪退！issue #579

### Reproduce method/复现方法

修补我的boot.img

### Expected behavior/预期行为

修补成功，输出镜像大小不为0

### Actual behaviour /实际行为

输出镜像大小为0

### Screenshots/截图

[image]

### Logs/日志

[APatch_bugreport_2024-08-02_18_16.tar.gz](https://github.com/user-attachments/files/16469176/APatch_bugreport_2024-08-02_18_16.tar.gz)

### Device Name/设备名称

PD2057 (Vivo Y52s)

### OS Version/系统版本

Android 10

### APatch Version/APatch 版本

10763

### Kernel Version/内核版本

Linux localhost 4.14.141+ #1 SMP PREEMPT Thu Nov 30 17:24:16 CST 2023 aarch64 Android

### KernelPatch Version/KernelPatch 版本

N/A

### Additional context/其他信息

N/A

closed 2024-08-02 by marat2509

a1489632, 2024-08-02:
> > * If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
> 
> From patch logs:
> 
> > ```
> > [?] no linux_banner? maybe CONFIG_CALLSYMS_ALL=n?
> > ```
> 
> Read https://github.com/bmax121/APatch#requirement

```
~ # zcat /proc/config.gz|grep CONFIG_KALLSYMS
CONFIG_KALLSYMS=y
# CONFIG_KALLSYMS_ALL is not set
# CONFIG_KALLSYMS_ABSOLUTE_PERCPU is not set
# CONFIG_KALLSYMS_BASE_RELATIVE is not set
~ #
```
Are you kidding?
```
CONFIG_KALLSYMS=y and CONFIG_KALLSYMS_ALL=y

CONFIG_KALLSYMS=y and CONFIG_KALLSYMS_ALL=n: Initial support
```

marat2509, 2024-08-02:
> Are you kidding?

You didn't even specify which version of KernelPatch and APatch you used. However, I see that you used the latest release version, where the KPatch version is 0.10.7

There the support for CONFIG_CALLSYMS_ALL=n was only initial, and further improved with subsequent versions, but still unstable. 

Instead of swearing, download the latest nightly version of APatch, with the latest KernelPatch, which is what you should have checked when creating the issue here:

> ### Version requirement/版本要求
> * [ ]  I am using latest CI version of APatch/我正在使用最新 CI 版本

If the patch still fails, then create the issue in the repository with [KernelPatch](https://github.com/bmax121/KernelPatch). This is more relevant to it.