bmax121/APatch #597: Shamiko requires magisk canary >27005

306bobby-android, 2024-08-06:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

The current public release does not recognize the new zip format for Shamiko, and the recent beta release complains about magisk version being too low

### Reproduce method/复现方法

Attempt to install newest Shamiko by LSposed 

### Expected behavior/预期行为

Hopefully, canary support 

### Actual behaviour /实际行为

Lack of canary support 

### Screenshots/截图

[image]
 

### Logs/日志

Can add if needed 

### Device Name/设备名称

Pixel 5 

### OS Version/系统版本

crDroid 10.6 

### APatch Version/APatch 版本

10753 as well as 10907 

### Kernel Version/内核版本

4.19.294 

### KernelPatch Version/KernelPatch 版本

0.11.1 

### Additional context/其他信息

Many more spoofing/hiding modules are requiring magisk Canary, support may be worth looking in to

pomelohan, 2024-08-06:
It's none of APatch's business since Shamiko developers had refused to adapt for APatch and they were trying their best to prevent APatch user from using it.

closed 2024-08-06 by pomelohan

1q23lyc45, 2024-08-07:
You don't need to use Shamiko on Apatch. By default, Apatch is completely hidden for applications without authorized superusers.

In addition, under no circumstances is it recommended to use closed-source software that already has open source alternatives, such as Shamiko.Nowadays, the hiding effects of Kitsune Mask, KernelSU, and APatch are much better than Shamiko, and they are also open source.

306bobby-android, 2024-08-07:
I must be confused, the issue request is for magisk canary support

Shamiko is just the first I found that requires

zeromake, 2025-01-02:
手动解包，把版本号改掉安装上也不能正常启用，应该是用了 magisk 新 api 了