bmax121/APatch #604: APatch 10762 - Redmi K70 Ultra bootloop

Stardusstt, 2024-08-10:
### Please check before submitting an issue/在提交 issue 前请检查

- [X] I have searched the issues and haven't found anything relevant/我已经搜索了 issues 列表，没有发现于本问题相关内容
- [X] If patch failed, root failed, or device unable to boot after flashing the new boot.img. Please goto KernelPatch/修复失败或刷入修补后镜像不能启动，请前往 KernelPatch 提问
- [X] I will upload bugreport file in APatch Manager - Settings - Report log/我会上传 bureport 文件从 APatch 管理器 - 设置 - 发送日志
- [X] I know how to reproduce the issue which may not be specific to my device/我知道如何重新复现这个问题

### Version requirement/版本要求

- [X] I am using latest CI version of APatch/我正在使用最新 CI 版本

### Describe the bug/描述 bug

Redmi K70 Ultra bootloop with Mi logo.

### Reproduce method/复现方法

1. Installed 10762 apatch and patch boot.img
2. Flash via fastboot flash boot boot.img
3. reboot
4. Phone bootloop with Mi logo.

### Expected behavior/预期行为

Normal boot of system with APatched image

### Actual behaviour /实际行为

bootloop with Mi logo.

### Device Name/设备名称

Redmi K70 Ultra

### OS Version/系统版本

1.0.20.0.UNNCNXM

### APatch Version/APatch 版本

10762 

### Kernel Version/内核版本

6.1.57-android14-11-g2de3ba70f288-ab11985038

### KernelPatch Version/KernelPatch 版本

0.10.7

### Additional context/其他信息

-

supechicken, 2024-08-20:
你好，想問一下你是怎麼解bl鎖的，還是說還得要去小米社區老實答題？

Stardusstt, 2024-08-23:
> 你好，想問一下你是怎麼解bl鎖的，還是說還得要去小米社區老實答題？

花錢，畢竟沒辦法過實名

LolekLiam, 2024-10-27:
same issue here on Redmi 12 5G

closed 2025-03-09 by Admirepowered